dreemurrs-embedded/Pine64-Arch #313: Failing to wake screen

lllsondowlll, 2022-02-14:
- Device: Pinephone Pro
- Kernel Version : 5.16.9
- UI: Phosh

Note: This is not the same as the U-boot DMA suspend bug

### Steps to reproduce
Have working suspend via Tow-boot SPI / levinboot and initiate suspend

### Expected behavior
Screen to wake upon waking from suspend

### Actual behavior
Screen fails to wake after 10-15 minutes, audio can still be heard when moving the volume rockers. Sometimes the screen wakes after 1-2 minutes, othertimes it fails to wake altogether.

Also noticed an issue where this sometimes occur with suspend disabled entirely via Tweaks / Phosh settings only occured on latest update as of two days ago 02/12/22 - 02/14/22

Danct12, 2022-02-19:
I seem to have been unable to reproduce the issue on my PinePhone Pro. For me, it just woke up I can already unlock it instantly.

Maybe this issue only shows up if you use Tow-Boot? I'll try that.

lrustand, 2023-11-11:
I also have problems with waking the screen. The screen works fine on boot, and as long as I don't turn off the screen or allow it to sleep everything works fine, but once the screen turns off I have to reboot to make the screen come on again.