dreemurrs-embedded/Pine64-Arch #319: Question: Modem-Power Driver

Daedalusspacegames, 2022-02-17:
- Device: PinePhone 1.2
- Kernel Version : 5.16.8-1-danctnix
- UI: Plasma Mobile

Not a bug, merely a question about default configuration. While reading through documentation about installing Biktorgj's modem firmware, I noticed that the modem-power driver does not appear to be enabled. Upon further investigation, this driver does appear in the linux-megi kernel source, but is disabled as of this commit: b4e0dfa85c48a187302a8029c51d049e3d164b20
My question is, why was this change made, and is there any potential harm in an end user re-enabling it? 
Some clarification in documentation might be prudent as well, given that [the Wiki page on upgrading your modem's firmware within this repo](https://github.com/dreemurrs-embedded/Pine64-Arch/wiki/Upgrade-the-modem-firmware) appears to make reference to the modem-power driver as well.

Danct12, 2022-02-19:
Because we're now using [eg25-manager daemon](https://gitlab.com/mobian1/devices/eg25-manager).

You can enable it, but we'll not take any bug reports regarding to modem. And that'll also require you to recompile the kernel as well.

The wiki page is outdated, but the process to upgrade the modem should be the same.

Danct12, 2022-02-19:
We also have a forum over [PINE64's forum](https://forum.pine64.org/forumdisplay.php?fid=159). This is where discussions should go.

closed 2022-02-19 by Danct12