dreemurrs-embedded/Pine64-Arch #327: anbox PKGBUILD does not build

3nprob, 2022-02-22:
### Steps to reproduce
```
guzuta build --arch aarch64 --chroot-dir ./chroot-aarch64 --repo-name danctnix --repo-dir ./repo/danctnix/aarch64 --srcdest ./sources --logdest ./logs PKGBUILDS/danctnix/anbox
```

### Expected behavior
Package is built

### Actual behavior
Build  fails with compilation error

### Logfiles and additional information
```
:: Synchronizing package databases...
100%:: Starting full system upgrade...
 there is nothing to do
==> Making package: anbox 0_git20210625-2 (Tue Feb 22 18:17:03 2022)
==> Retrieving sources...
  -> Found anbox-04ac697c890b96aa70814a29696adee6a2fc7ff8.tar.gz
  -> Found cpu_features-b5c271c53759b2b15ff91df19bd0b32f2966e275.tar.gz
  -> Found sdbus_cpp-3b735bf1aad65277f56e65c828a22455cbaf5245.tar.gz
  -> Found inputmethod-hack.patch
  -> Found give-more-time-to-start.patch
  -> Found 0001-Don-t-display-Android-apps-in-the-app-drawer.patch
  -> Found 1827.patch
  -> Found anbox-container-manager.service
  -> Found anbox-session-manager.service
  -> Found 99-anbox.rules
  -> Found anbox.desktop
  -> Found dev-binderfs.mount
==> Validating source files with md5sums...
    anbox-04ac697c890b96aa70814a29696adee6a2fc7ff8.tar.gz ... Passed
    cpu_features-b5c271c53759b2b15ff91df19bd0b32f2966e275.tar.gz ... Passed
    sdbus_cpp-3b735bf1aad65277f56e65c828a22455cbaf5245.tar.gz ... Passed
    inputmethod-hack.patch ... Passed
    give-more-time-to-start.patch ... Passed
    0001-Don-t-display-Android-apps-in-the-app-drawer.patch ... Passed
    1827.patch ... Passed
    anbox-container-manager.service ... Passed
    anbox-session-manager.service ... Passed
    99-anbox.rules ... Passed
    anbox.desktop ... Passed
    dev-binderfs.mount ... Passed
==> Making package: anbox 0_git20210625-2 (Tue 22 Feb 2022 06:17:20 PM UTC)
==> Checking runtime dependencies...
==> Installing missing dependencies...
resolving dependencies...
looking for conflicting packages...
warning: dependency cycle detected:
warning: mesa will be installed before its libglvnd dependency

Packages (38) giflib-5.2.1-2  hidapi-0.11.2-1  libdrm-2.4.110-1  libedit-20210910_3.1-1  libglvnd-1.4.0-1  libibus-1.5.25-5  libjpeg-turbo-2.1.2-1  libomxil-bellagio-0.9.3-3
              libpciaccess-0.16-2  libpng-1.6.37-3.1  libtiff-4.3.0-1  libunwind-1.6.2-1  libusb-1.0.25-2  libwebp-1.2.2-1.1  libx11-1.7.3.1-1  libxau-1.0.9-3  libxcb-1.14-1
              libxcursor-1.2.0-2  libxdamage-1.1.5-3  libxdmcp-1.1.3-3  libxext-1.3.4-3  libxfixes-6.0.0-1  libxrender-0.9.10-4  libxshmfence-1.3-2  libxxf86vm-1.1.4-4
              llvm-libs-13.0.1-1  lm_sensors-1:3.6.0.r41.g31d1f125-1  mesa-21.3.6-1  sdl2-2.0.20-2  vulkan-icd-loader-1.2.203-1  wayland-1.20.0-1  wget-1.21.2-1  xcb-proto-1.14.1-5
              xorgproto-2021.5-1  boost-libs-1.78.0-1  lxc-1:4.0.12-1  protobuf-3.19.4-1  sdl2_image-2.0.5-2

Total Installed Size:  346.18 MiB

:: Proceed with installation? [Y/n]
(38/38) checking keys in keyring                                                                             [#################################################################] 100%
(38/38) checking package integrity                                                                           [#################################################################] 100%
(38/38) loading package files                                                                                [#################################################################] 100%
(38/38) checking for file conflicts                                                                          [#################################################################] 100%
(38/38) checking available disk space                                                                        [#################################################################] 100%
:: Processing package changes...
( 1/38) installing wget                                                                                      [#################################################################] 100%
Optional dependencies for wget
    ca-certificates: HTTPS downloads [installed]
( 2/38) installing lxc                                                                                       [#################################################################] 100%
[... 251 lines cut ...]
>’?
   18 | #include <boost/filesystem.hpp>
  +++ |+#include <memory>
   19 |
/build/anbox/src/anbox-04ac697c890b96aa70814a29696adee6a2fc7ff8/external/xdg/xdg.cpp:179:44: error: no declaration matches ‘std::shared_ptr<xdg::BaseDirSpecification> xdg::BaseDirSp
ecification::create()’
  179 | std::shared_ptr<xdg::BaseDirSpecification> xdg::BaseDirSpecification::create()
      |                                            ^~~
/build/anbox/src/anbox-04ac697c890b96aa70814a29696adee6a2fc7ff8/external/xdg/xdg.cpp:179:44: note: no functions named ‘std::shared_ptr<xdg::BaseDirSpecification> xdg::BaseDirSpecifi
cation::create()’
In file included from /build/anbox/src/anbox-04ac697c890b96aa70814a29696adee6a2fc7ff8/external/xdg/xdg.cpp:16:
/build/anbox/src/anbox-04ac697c890b96aa70814a29696adee6a2fc7ff8/external/xdg/xdg.h:90:7: note: ‘class xdg::BaseDirSpecification’ defined here
   90 | class BaseDirSpecification : NotCopyable, NotMoveable
      |       ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [external/xdg/CMakeFiles/xdg.dir/build.make:76: external/xdg/CMakeFiles/xdg.dir/xdg.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:672: external/xdg/CMakeFiles/xdg.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
==> ERROR: Build failed, check /home/user/dev/Pine64-Arch/chroot-aarch64/user/build
Error: Unable to build package in PKGBUILDS/danctnix/anbox

Caused by:
    makechrootpkg failed
```

3nprob, 2022-02-22:
Looks like this issue appears when building with glibc >= 2.34 and fixed in https://github.com/anbox/anbox/commit/4ef88ab7005733480e88a4aab71de23f27c49842

Danct12, 2022-02-23:
Fixed in c0e8a40e30260fc2fd9085a0edead270b776e14c

closed 2022-02-23 by Danct12