dreemurrs-embedded/Pine64-Arch #332: Sound not working

Schmiddiii, 2022-02-26:
- Device: PinePhone
- Kernel Version : 5.16.10-1-danctnix
- UI: Phosh

### Steps to reproduce

Base Install of the OS.

### Expected behavior

Sound to be working 

### Actual behavior

No sound devices seem to be found (Settings -> Audio shows no devices in in/output), see logs.

### Logfiles and additional information

`aplay -l` (Translated)

```
**** List of Hardware-Devices (PLAYBACK) ****
Card 0: PinePhone [PinePhone], Device 0: 1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0 [1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0]
  Sub-Devices: 0/1
  Sub-Device #0: subdevice #0
```

`pactl list` (Translated)

```
Connectinoerror: Connection denied
pa_context_new() failed: Connection denied
```

`journalctl` (Only parts, the real is way to long, included as attachment, otherwise comment would be too long)
[journalctl-shorteded.log](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8147467/journalctl-shorteded.log)

Danct12, 2022-02-27:
The log gives somewhat a interesting result:
"/usr/lib/gstreamer-1.0/gst-plugin-scanner: Zeile 1: Syntaxfehler beim unerwarteten Symbol"

Translated from German:
"/usr/lib/gstreamer-1.0/gst-plugin-scanner: Line 1: Unexpected icon syntax error"

I wonder why, maybe corruptions?

Schmiddiii, 2022-02-27:
I just reinstalled the OS again and it was fixed. Seemed to just have been a corruption.

closed 2022-02-27 by Schmiddiii