dreemurrs-embedded/Pine64-Arch #336: Copy to and paste from the clipboard

kclisp, 2022-03-04:
- Device: PinePhone 
- Kernel Version : 5.16.10-1-danctnix
- UI: Phosh

What is the current expected way to copy to and paste from the clipboard, in say the terminal or chatty? A long press on a selection doesn't do anything. In the terminal, Ctrl + C/V and Ctrl + Shift + C/V don't do anything. The only way I've found to copy is to use the terminal keyboard, press >_, press menu, and press copy.

Is there a package or something that helps with the clipboard (which can perhaps be installed by default)?

mikeshenson, 2022-03-04:
In the terminal, a Long press will create a popup with paste and select all.
In chatty, I use the CTRL + V on the keyboard.  (I created a custom layout[1] to have access to CTRL on the letter pad)

[1] https://github.com/mikeshenson/Phosh-Squeekboard-Keyboard

kclisp, 2022-03-04:
Right, so is the expected methods as follows?
- Terminal
  - Copying: >_ -> menu -> copy
  - Pasting: Long press or >_ -> menu -> paste
- Chatty
  - Copying: >_ -> menu -> copy
  - Pasting: Ctrl + V or >_ -> menu -> paste

It seems like the current most uniform way is to use copy and paste from >_ -> menu in the terminal keyboard.

A more intuitive method would be the ability to copy a selection from long press, and paste from long press. (Or for the terminal, Ctrl + Shfit + C/V.)