dreemurrs-embedded/Pine64-Arch #349: Initial (waking) SMS not available when waking from suspend

fbmoose48, 2022-03-17:
- Device: PinePhone Pro
- Kernel Version : 5.16.10-1-danctnix
- UI: Phosh

### Steps to reproduce
any progress in suspend/wake-from-suspend on PPP is great. However, it seems that while an initial text will wake the phone up to receive additional texts, the first text is not actually received and displayed by Chatty, mmcli, or ModemManagerGUI - as though it was never actually cached by the modem.

### Expected behavior
ALL texts received and displayed

### Actual behavior
initial text not available, additional texts received after wake available

### Logfiles and additional information

ArenM, 2022-03-21:
Possibly related: https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/466

fbmoose48, 2022-03-22:
Similar to that post I'm also on Verizon CDMA network, however, I'm using stock modem FW on the PPP. I `changed ATTR{power/control}="on"` instead of `"auto"`  in `/usr/lib/udev/rules.d/80-modem-eg25.rules ` as suggested in that post, all it did was make my phone not wake at all and need to be rebooted after suspend. Also, I don't have a modem reintialization issues after suspend like OP, mine is seems more like modem reinitializes too slow, than Chatty doesn't re-poll the modem for messages received while suspended after the 3-5 second wakeup process.

fbmoose48, 2022-03-30:
Is there a recommended version of u-boot or tow-boot that might resolve this?

fbmoose48, 2022-04-14:
any resolution? Is this unique to CDMA carriers?