dreemurrs-embedded/Pine64-Arch #352: unable to send/receive SMS

fbmoose48, 2022-03-18:
- Device: PinePhone Braveheart
- Kernel Version : 5.16.14-1-danctnix
- UI:  phosh

### Steps to reproduce
can't send/receive SMS, but can place/receive calls

### Expected behavior
SMS work

### Actual behavior
SMS not working

### Logfiles and additional information
I had faced SMS issues related to modem cache in the past, but had been able to manage by by connecting to the modem tty through minicom and clearing cache with "AT+CMGD=0,4". Now that doesn't seem to work anymore.

Danct12, 2022-03-23:
IIRC, this might had to do with 3G/4G thing, try switch to 3G and see if you can receive SMS.

Danct12, 2023-09-07:
Closing as the reporter hasn't responded to the issue for a year now.

closed 2023-09-07 by Danct12