dreemurrs-embedded/Pine64-Arch #353: Waydroid only works with hardward acceleration disabled

DSchroer, 2022-03-18:
- Device: Pinephone Pro
- Kernel Version : 5.16.14-1-danctnix
- UI: Phosh

### Steps to reproduce
1. install waydroid: `sudo pacman -S waydroid`
2. setup images: `sudo waydroid init`
3. start service `sudo systemctl enable waydroid && sudo systemctl start waydroid`
4. launch UI `waydroid show-full-ui`

### Expected behavior
Waydroid UI Opens

### Actual behavior
The UI tries to open but the screen becomes glitchy and unresponsive.

Had to disable hardware acceleration first: `( sudo echo "ro.hardware.gralloc=default" ; sudo echo "ro.hardware.egl=swiftshader" ) >>/var/lib/waydroid/waydroid_base.prop`

Now the UI loads but is slow.

e-dorigatti, 2022-03-23:
It's not an arch-specific issue as I have the same issue with PostmarketOS+Phosh/PlasmaMobile on the Pinephone Pro, while everything is fine on the non-pro (arch+phosh).

DSchroer, 2022-03-23:
If that's the case which project should this be referred to?

Arsen6331, 2022-03-31:
I've already opened an issue for this: waydroid/waydroid#291

rothn, 2022-09-19:
Building from latest repo sources seems to solve the problem: [guide](https://docs.google.com/document/d/1Sly_VH3w6wFIdE72WXijQegoHZh8IxEnJ9m0hH7GodU/edit#)

closed 2023-04-12 by Danct12