dreemurrs-embedded/Pine64-Arch #355: Plasma Mobile: Call audio not rerouted

emulti, 2022-03-20:
- Device: PinePhone
   - Kernel Version : 5.16.13-1-danctnix
   - UI: Plasma Mobile

### Steps to reproduce

Outgoing call: recipient can pick up (Plasma dialer UI still says "Calling..." but no audio in either direction
Incoming Call: PP can pick up, but no audio in either direction

### Expected behavior
Calls work with audio rerouted to the "Voicecall" Alsa profile when connection is made.

### Actual behavior
No Audio rerouting takes place.
In journalctl -f kde-telephony-daemon announces "Skipping audio setup because no voice compatible card found"
Traced to: https://invent.kde.org/plasma-mobile/plasma-dialer/-/blob/master/kde-telephony-daemon/src/dialer-audio.cpp (L104)
But pactl list shows the Mic sources and Speaker/Earphone sinks

jsarkari01, 2022-03-24:
I have found that when I call my PPP I can hear on both mobiles. But when I call from PPP I cannot hear audio in speaker or in earpiece on PPP only but I can from my other mobile.

Danct12, 2024-09-28:
I'll be closing this issue as this is an old issue and may no longer be an issue. Please reopen if you can reproduce the issue.

closed 2024-09-28 by Danct12