dreemurrs-embedded/Pine64-Arch #357: Autodetected APN for Telia network which worked and found it - but after browsing the web the modem disconnected

jsarkari, 2022-03-24:
- Device: (PinePhone or PineTab?) PinePhone Pro
- Kernel Version : $(uname -r) 5.16.15-1-danctnix
- UI: KDE Plasma Version: 5.24.3

### Steps to reproduce

Found that I needed to check APN as no mobile data currently.
1, Settings
2. Cellular Network
3. Enter password (pin)
4. APN
5. Selected SIMs - SIM 0 (Might be an idea to increase the number by 1)
6. APNs: - Modify Access Point Names
7. Autodetect
8. Found the network and the correct one was already selected for my network (Telia in Sweden)

### Expected behavior

It found the APN for my network and some others but was already selected for mine.
Test browsing the web etc sucessfully.

### Actual behavior

Opened Angelfish and tried to open a few web pages then after a minute or so the Modem disconnected.
Then I got a message popping up  "System policy prevents controlling the Modem Manager"
So entered my pin and the modem was still disconnected.
Only way to recover is to reboot.
But its an endless loop with this problem.

### Logfiles and additional information

Stianlyng, 2023-03-13:
@jsarkari  Do you mind sharing the APN you found? I`m struggeling a bit. Was it telia.cmg ?

jsarkari, 2023-03-13:
I don't have that mobile anymore, so you can close if you wish

closed 2023-03-13 by jsarkari