dreemurrs-embedded/Pine64-Arch #361: Regression with linux-megi-5.17: screen freezing in Waydroid

dariox86, 2022-04-02:
- Device: PinePhone
- Kernel Version : 5.17.0-2-danctnix
- UI: Phosh

### Steps to reproduce
Upgrade from linux-megi-5.16.15-1 to linux-megi-5.17.0-2.

### Expected behavior
Open WhatsApp on Waydroid and use it with no issues.

### Actual behavior
Upon sending a message to someone and going back to the contact list the screen freezes. It is still possible to close Waydroid and the application is actually running in background even though the display is not updated. It should be possible to reproduce the issue with other applications as well after a bit of fiddling.

### Additional notes
I experienced lack of mobile Internet connectivity upon boot for several minutes. An orange exclamation mark showed up on the mobile signal icon, though I am unsure whether this is related to the new kernel. It only happened twice.

Danct12, 2022-04-07:
This appears to be a specific issue with WhatsApp as someone else has tested it and the issue is there for them. I don't use WhatsApp so I can't check.

whipplej5, 2022-04-07:
I played an android game called Ticket to Ride on Waydroid that actually worked quite well until the kernel update. Now it freezes intermittently and is pretty much unusable.

That is not a big deal for the overall usability of the phone/OS, but I do think that it may be a sign that the kernel change may affect more than WhatsApp

closed 2022-12-06 by dariox86