dreemurrs-embedded/Pine64-Arch #363: Unable to add a print in Arch Linux phosh

jsarkari, 2022-04-04:
- Device: (PinePhone or PineTab?) PinePhone Pro
- Kernel Version : $(uname -r) 5.17.0-2-danctnix
- UI: phosh

### Steps to reproduce

Open settings, printers
Add a printer

### Expected behavior

Ability to add a printer on wifi etc

### Actual behavior

Sorry! The system printing service doesn't seem to be available.

### Logfiles and additional information

Danct12, 2022-04-07:
The error can be fixed by installing `cups` and enable the service, I'll enable this for the next image.

jsarkari, 2022-04-07:
I installed cups:

sudo pacman -S cups

So I can now add a printer via wifi or usb directly however I then got a system notification:

Additional Printer Drivers Required
An application is reqesting additional printer drivers.

Find in Software

This is an EPSON ET-2750 printer.

Danct12, 2022-04-07:
Hi, it looks like your printer requires a driver (see [Arch Wiki](https://wiki.archlinux.org/title/CUPS/Printer-specific_problems#Epson)).

Please try to build and install the driver for your specific printer (for your printer, it should be [epson-inkjet-printer-escpr](https://aur.archlinux.org/packages/epson-inkjet-printer-escpr) from AUR), this might make the printer working.

jsarkari, 2022-04-07:
Yes I am doing some research to resolve the issues. I have now managed to add the ppd file via localhost:631 so getting closer.

Danct12, 2022-05-08:
This is now included in Phosh and Plasma images. I'll be closing this as this seems to be a issue with your printer model under Linux.

closed 2022-05-08 by Danct12