dreemurrs-embedded/Pine64-Arch #364: Unable to setup vpn via PIA OVPN exported files

jsarkari, 2022-04-05:
- Device: (PinePhone or PineTab?) PinePhone Pro
- Kernel Version : $(uname -r) 5.17.0-2=danctnix
- UI: phosh

### Steps to reproduce

Open Settings
Network
VPN - Select +
Select - Import from file...
Select file to import (...ovpn)

### Expected behavior

Import the VPN file successfully

Note: I tried the same files in Manjaro ARM phosh and they imported successfully and was able to activate VPN in different countries.

One thing I did see in the other OS was an option for openvpn which is not implemented here in Arch Linux ARM phosh dev yet.

### Actual behavior

Message:

Cannot import VPN connection

The file "sweden.ovpn" could not be read or does not contain recongnised VPN connection information

Error: unknown error.

### Logfiles and additional information

Danct12, 2022-04-07:
Try install `networkmanager-openvpn` and let me know if that fixed it.

jsarkari, 2022-04-07:
So I have tried to install it using:

pacman -Q networkmanager-openvpn

error: package 'networkmanager-openvpn' was not found

I also tried with sudo ....

Same error

I then checked version of networkmanager installed:

pacman -Q networkmanager

networkmanager 1.36.4-1

jsarkari, 2022-04-07:
I then tried installing with -S and it worked.

jsarkari, 2022-04-07:
This is now working and I can import various openvpn files and switch between those in the VPN section in Network.

closed 2022-04-07 by jsarkari