dreemurrs-embedded/Pine64-Arch #368: phosh.service fails (Pinetab)

RightToPrivacy, 2022-04-06:
- Device:  Pinetab
- Kernel Version : 5.17
- UI: Phosh

### Steps to reproduce

Upgrading latest Kernel + latest Phoc package, reboot and problem occurs.

### Expected behavior

Expected to boot into graphical Phosh interface.

### Actual behavior

After decrypting / , Danctnix Arch logo + blank screen persists

### Logfiles and additional information

Problems appear during failure of phosh.service (networking even begins etc):

Apr 04 18:21:46 laptop phoc[3422]: Failed to parse config /etc/phosh/phoc.ini: Key file contains line “map-to-output:DSI-1” which is not a key-value pair, group, or comment
Apr 04 18:21:46 laptop systemd-logind[2963]: Session 11 logged out. Waiting for processes to exit.
Apr 04 18:21:46 laptop systemd[1]: session-11.scope: Deactivated successfully.
Apr 04 18:21:46 laptop systemd[1]: phosh.service: Main process exited, code=exited, status=1/FAILURE
Apr 04 18:21:46 laptop systemd[1]: phosh.service: Failed with result 'exit-code'.
Apr 04 18:21:46 laptop audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=phosh comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Apr 04 18:21:46 laptop kernel: audit: type=1131 audit(1649110906.700:138): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=phosh comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Apr 04 18:21:46 laptop systemd-logind[2963]: Removed session 11.

(can include more log if needed but condensed for space - shutting down starts after the failure of phosh.service and failure to parse /etc/phosh/phoc.ini)

Danct12, 2022-04-07:
Looks like the error is here:

```
Apr 04 18:21:46 laptop phoc[3422]: Failed to parse config /etc/phosh/phoc.ini: Key file contains line “map-to-output:DSI-1” which is not a key-value pair, group, or comment
```

Try remove (or move it to somewhere else) /etc/phosh/phoc.ini and see if it works.

RightToPrivacy, 2022-04-07:
I did try some minor commenting out of the line yesterday (just to see what would happen in logs) but had no luck.

Update: after a downgrade (to phoc 0.12.0-1) Pinetab appears to be booting into phosh again (only time it has in a couple days - have been trying things from sdcard Arch).

Thank you for your response. :)

Danct12, 2022-04-07:
I highly recommend to talk to people at `#phosh:talk.puri.sm` on Matrix.

RightToPrivacy, 2022-04-07:
Update: I upgraded to latest phoc 0.13.1-1. Fix (for me) for anyone sharing this experience: remove the original/old /etc/phosh/phoc.ini, replacing it by copying the (different) /usr/share/phoc.ini

Then edit the line:

[output:DSI-1]
scale = 2

Change the above scale = 2 to scale = 1 to restore original scale size. Everything seems to work okay now in relation to this issue.

I'll close with this comment to get rid of issue. 

Thanks again!

closed 2022-04-07 by RightToPrivacy