dreemurrs-embedded/Pine64-Arch #373: Timezone not set up by default

emulti, 2022-04-10:
- Device: Pinephone
- Kernel Version : 5.17.0-2-danctnix
- UI: phosh
- Based on 2022024 image with updates

### Steps to reproduce
After UK move to daylight savings time, Phosh continues to display GMT rather than BST (British Summer Time), GMT+1

### Expected behavior
Displayed system time should update automatically

### Actual behavior
GMT continues to be used
Settings/Date and Time has 'Automatic Date and Time' and 'Automatic Time Zone' set to 'on'

### Fix/Workaround
Set up /etc/localtime symlink as in Arch install guide
In the case of London timezone: 
`$ sudo ln -sf /usr/share/zoneinfo/Europe/London /etc/localtime`

Correct local time is now displayed in Phosh. When travelling to a different timezone, however, manual update of Timezone will be needed.
Please add a note to install guide to this effect.

Danct12, 2023-09-07:
Is this still an issue? I find that's likely to be a systemd timezone issue than a Phosh/GNOME issue. If not, please close the issue. Thanks!

emulti, 2023-09-07:
Presumably still an issue because time is set by NTP rather than the Cellular Network like commercial OS phones. So timezone will have to be updated when user travels in order to correctly show local time.

closed 2023-09-07 by emulti