dreemurrs-embedded/Pine64-Arch #376: Wrong keymap in latest linux-megi kernel (5.17.2-1)

Dan-Sun, 2022-04-17:
- Device: PinePhone
- Kernel Version : 5.17.2-1-danctnix
- UI: Phosh

I actually just wanna keep issue-#304 running/updated (info) until the keyboard, including keymap, is fully working via kernel.

ragreenburg, 2022-04-21:
If you're talking about the function keys working instead of the special characters next to them, that seems to be by design, according to Megi. I also ended up downgrading my kernel as well and blocked it from updating until I could figure out how to get a custom keymap working.

tanabi, 2022-04-21:
You guys have probably already seen this, but I'll throw it out in case you haven't.  Megi noted this re. making keymaps:

https://xnux.eu/log/#066

When he refers to XKB, https://help.ubuntu.com/community/Custom%20keyboard%20layout%20definitions has some information.

Not sure if this helps anyone, sorry if it is old news :)

ragreenburg, 2022-04-21:
@tanabi Thanks for the XKB like, that's really helpful! I will have to test it out with the keyboard.

sado1, 2022-05-14:
https://github.com/dreemurrs-embedded/Pine64-Arch/commits/master/PKGBUILDS/danctnix/xkeyboard-config
I think these changes were supposed to help (adding proper keymap for the function/number key row), but on fully updated PlaMo build, I am unable to use the third row with any modifier key I can think of. (I am not sure this is fully configured yet)

JonathanReeve, 2022-05-22:
It seems like this problem persists with linux-megi-5.17.6-1. The workaround isn't working for me. This could be related to #296. 

Trying to type symbols on the top row, with FN+1, FN+2, etc., either produces no output, or the wrong output. FN+1, FN+2, produce no output, while other numbers all print `~`. This is exceptionally annoying, since I can type commands in a terminal, for instance, but can't type `-`, which is necessary for just about every terminal command. 

@Danct12 , could it be that you have the wrong keycodes in [this commit](https://github.com/dreemurrs-embedded/Pine64-Arch/commit/7cb68bf1d1cf33fbec02800e94740f35ca93f269)?

JonathanReeve, 2022-05-22:
[Related post about this on the Pine forums. There are some pretty hacky workarounds, but no fix, yet](https://forum.pine64.org/showthread.php?tid=16521)

Dan-Sun, 2022-05-23:
I haven't tried it, but here is maybe another solution. I saw it in the pine64 forum.
https://codeberg.org/HazardChem/PinePhone_Keyboard

It may also work on phosh.

sado1, 2022-05-26:
> > I haven't tried it, but here is maybe another solution. I saw it in the [pine64 forum](https://forum.pine64.org/showthread.php?tid=16521&pid=110022#pid110022). https://codeberg.org/HazardChem/PinePhone_Keyboard
> > It may also work on phosh.
> 
> Is there an easy / ELI5 explination on how to do this?

As the newest danctnix build already has the needed xkb configuration, follow the instructions at the end of https://wiki.mobian-project.org/doku.php?id=ppaccessories - you simply need to add the override file and reboot or restart Phosh.  Then, AltG can be used to use the top row's keys that were missing earlier.

Dan-Sun, 2022-05-27:
That's great!  That's a way better solution. Thanks for sharing @sado1 ! 

```
systemctl edit phosh.service
```
```
[Service]
Environment=XKB_DEFAULT_MODEL=ppkb
```
> After this, the top row symbols will be available as L3-{0-9}. The L3 key can be configured in settings...

`Settings → Keyboard → Special Character Entry -> Alternate Characters Key`

hal9yo, 2022-05-29:
> As the newest danctnix build already has the needed xkb configuration, follow the instructions at the end of https://wiki.mobian-project.org/doku.php?id=ppaccessories - you simply need to add the [...]

So you don't need to install xkb-data? Just set the environment variable "XKB_DEFAULT_MODEL=ppkb" ?

I got confused because I tried installing xkb-data and didn't find it in the repos, and then I was like, am I supposed to build it from source?

I'm running the SXMO build, I dunno if that makes a difference, is anyone doing this on the SXMO build?

Crsarmv7l, 2022-05-30:
@Dan-Sun thanks for typing the specifics out. I had tried the mobian instructions before with no luck. I had to use sudo for the systemctl but it works great now.

I posted your comment in the Arch pinephone forum (top thread is this issue), with credit to you and sado1 (who also posteed there!)

kuba160, 2022-06-10:
@hal9yo :
> I'm running the SXMO build, I dunno if that makes a difference, is anyone doing this on the SXMO build?

You currently have to add this to your sway config file:
```
input "0:0:PinePhone_Keyboard" {
      xkb_layout us
      xkb_model ppkb
      xkb_options lv3:ralt_switch
}
```
Here you can find issue discussed in postmarketOS: https://gitlab.com/postmarketOS/pmaports/-/issues/1515

It also looks like the fix has been added to sxmo itself, but it is not available in 1.9.0: https://git.sr.ht/~mil/sxmo-utils/commit/d6a6fc2c4767a49e8fd1be7293c2bbd47da3f811

I am not liking the new layout so much, really enjoyed writing minus sign with Fn + 9. I also wanted to use right alt as shift for letters. I found a way to achieve this by adding these lines to ppkb symbols in `/usr/share/X11/xkb/symbols/inet`:

<details>
  <summary>ppkb symbols I used:</summary>

```
    key <FK08>   {       [ equal              ] }; // F8
    key <FK09>   {       [ minus, underscore  ] }; // F9
    key <FK10>   {       [ plus, equal        ] }; // F10

    key <LatQ> { [ q,Q,Q] };
    key <LatW> { [ w,W,W] };
    key <LatE> { [ e,E,E] };
    key <LatR> { [ r,R,R] };
    key <LatT> { [ t,T,T] };
    key <LatY> { [ y,Y,Y] };
    key <LatU> { [ u,U,U] };
    key <LatI> { [ i,I,I] };
    key <LatO> { [ o,O,O] };
    key <LatP> { [ p,P,P] };
    key <LatA> { [ a,A,A] };
    key <LatS> { [ s,S,S] };
    key <LatD> { [ d,D,D] };
    key <LatF> { [ f,F,F] };
    key <LatG> { [ g,G,G] };
    key <LatH> { [ h,H,H] };
    key <LatJ> { [ j,J,J] };
    key <LatK> { [ k,K,K] };
    key <LatL> { [ l,L,L] };
    key <LatZ> { [ z,Z,Z] };
    key <LatX> { [ x,X,X] };
    key <LatC> { [ c,C,C] };
    key <LatV> { [ v,V,V] };
    key <LatB> { [ b,B,B] };
    key <LatN> { [ n,N,N] };
    key <LatM> { [ m,M,M] };
```
</details>

Also I used `xkb_options lv3:ralt_switch_multikey` in sway config which makes Shift + Right Alt act as compose key. I wish I would be able to use Page Up/Down as Ctrl + Home/End but I don't know yet how to achieve this.

bakerk98, 2022-07-04:
Any advice on how to set this in Plasma Mobile?