dreemurrs-embedded/Pine64-Arch #377: Pinephone Pro File system mount as read only when resuming from Suspend

mibry, 2022-04-21:
- Device: PinePhone Pro with Tow boot install on SPI
-  Kernel Version : 5.17.3-1-danctnix
- UI: SXMO

### Steps to reproduce
Boot to SXMO on emmc put pinephone pro into suspend for 20 mins. Wake the phone and try to excuute command. Command returns input output error. When when try to run dmesg input out put error report and files system mount as read only reported in standard out.

### Expected behavior
Expected Phone to exit suspend and be usable.

### Actual behavior
As above

### Logfiles and additional information
NA

Danct12, 2022-04-21:
You can use netcat to get the log from the device.

Alternatively, dump the log into /tmp and use scp to retrieve it.

mibry, 2022-04-21:
I captured dmesg and journalctl let me know you need more
[dmesgout.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8530706/dmesgout.txt)
[journalctloutput.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8530712/journalctloutput.txt)
[journalctloutput1.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8530714/journalctloutput1.txt)

mibry, 2022-04-22:
I am closing the issue. Looks like it might be a hardware problem. I saw that @alexmitter on telegram had the same problems.

closed 2022-04-22 by mibry

Danct12, 2022-04-22:
From alexmitter:

> [ 46.453059] mmc2: mmc_select_hs200 failed, error -110
[ 46.458745] mmc2: error -110 doing runtime resume

>Yep, same issue as with my borked unit. I now own another PPP that works flawlessly with exactly the same software on it, so I suspect some hardware issue.

>Usual symptoms of it are that mounting root on boot sometimes fails and resuming the emmc from suspend fails.

Danct12, 2022-04-23:
Same issue reported by a user on postmarketOS bug tracker: https://gitlab.com/postmarketOS/pmaports/-/issues/1500