dreemurrs-embedded/Pine64-Arch #378: alsa-card-profiles 1:0.3.50-1 breaks call audio (PPP)

fbmoose48, 2022-04-21:
- Device: PinePhone Pro
- Kernel Version : linux-megi (5.17.0-2)
- UI:  phosh

### Steps to reproduce
Place call, no audio/mic

### Expected behavior
audio profile for calls works

### Actual behavior
silence

### Logfiles and additional information
downgrading package alsa-card-profiles (1:0.3.50-1 => 1:0.3.49-1) fixed issue

Danct12, 2022-04-22:
I'll investigate this, for now, I've added 0.3.49 to the repo so it won't auto update to it.

greg-hellings, 2022-05-04:
Downgrading did not give audio in calls. I did not have it before the upgrade either. Including with the most recent release clean installed. Audio works fine outside of call configuration.

Danct12, 2022-06-01:
All of this might be related to this: https://archlinux.org/news/undone-replacement-of-pipewire-media-session-with-wireplumber/

I'm currently running the latest image and updated everything, still works.

closed 2022-06-01 by Danct12