dreemurrs-embedded/Pine64-Arch #381: Watching videos on websites in Firefox eg: news videos are playing at 3-4 times the speed

jsarkari, 2022-04-26:
- Device: (PinePhone or PineTab?) Pinephone Pro
- Kernel Version : $(uname -r) 5.17.4-1-danctnix
- UI: phosh

### Steps to reproduce

Open firefox
Go to bbc.com/news
Open an article where there is an embedded video
Start playing the video

### Expected behavior

Plays at normal speed so you can comprehend what is being said

### Actual behavior

It plays at 3-4 times the speed so they sound like mice!

Furthermore there is no option in video playback to change any settings like speed, adding subtitles etc

### Logfiles and additional information

jsarkari, 2022-04-26:
> This is a known bug with the PinePhone Pro, try suspend the device and resume again.

Is there a way to force suspend rather than waiting the minimum 5 minutes?

Danct12, 2022-04-26:
Yes, run `systemctl suspend` in the terminal.

jsarkari, 2022-04-26:
Yes found that and it worked, odily it was also doing it in chromium browser also.

Danct12, 2022-04-26:
That is because it's not a Chromium or any program issue, but rather a issue in the audio driver itself.

GertVanKruiningen, 2022-06-23:
Suspending using "systemctl suspend" or waiting for the PPP to suspend, does fix the audio issue, but doesn't always work.
Suspending switches between "fast" and "normal" speed, but no idea yet why its not consistent.

jsarkari, 2022-06-23:
I could always record the songs and sell as remixes!

caveman250, 2022-07-06:
Is there an issue or thread that can be followed for the driver bug? would love to find out more, especially as to why suspending only fixes the issue sometimes.

dukla2000, 2022-08-28:
I still have this issue with 5.19.3-1
Re-creatable by installing archlinux-pinephone-pro-phosh-20220729.img
edit /etc/pacman.conf and enable testing repo
pacman -Syu

Reboot, allow to suspend after 5 mins. Push power button - PPP comes back to life. Any sound (e.g. ping when plugging in power) actually results in about 10 second of "short wave radio" (this has changed from Mickey Mouse at earlier kernel revs.)

I have never been able to get multiple suspend/resume cycles to give me any different result.

I can't see anything peculiar (to my eye) in the journal worth reporting

Danct12, 2022-09-28:
I'll be closing this issue for now since a lot of people has this problem fixed on their end with the latest kernel.

closed 2022-09-28 by Danct12