dreemurrs-embedded/Pine64-Arch #385: PPP keyboard charging very slowly if at all

jsarkari, 2022-05-04:
- Device: (PinePhone or PineTab?) PPP
- Kernel Version : $(uname -r) 5.17.5-1-danctnix
- UI: phosh

### Steps to reproduce

So I have left the charger in the keyboard, pressed the keyboard button so the Main is showing as charging and the Extra is not charging.

But it hardly charges hovering around 1%/2%? I only had this keyboard a few days and not been able to charge it fully yet?

### Expected behavior

Plug in the charger and be able to charge the keyboard correctly.

### Actual behavior

It is hardly charging at all. Having used the standard charger that I use for the PPP which charges at the normal speed.

### Logfiles and additional information

Danct12, 2022-05-04:
Is it USB-A and how many amps?

jsarkari, 2022-05-04:
It is USB-C and max 3.0amps

jsarkari, 2022-05-04:
I saw on Discord from: input current still needed.  my keyboard is recharging not the main battery.  the labels are flipped when the keyboard is attached

I have also checked using the following to see if the keyboard is actually charging: 
cat /sys/class/power_supply/ip5xxx-battery/status
Charging

cat /sys/class/power_supply/ip5xxx-battery/current_now
1492

I have also tried a 100W double charger  and the current_now is 5968 but it is hardly charging.

Currently the Main battery is at 1% most of the time now.

Danct12, 2022-05-04:
I believe the PinePhone keyboard is not USB-PD compliant. Please try a dumb charger brick and see if that works.

tanabi, 2022-05-04:
The PPP is incredibly, stupidly picky about chargers.  Mine won't charge up on most of my chargers that will quick-charge other phones.  Also mine won't charge using my Thinkpad's USB-C power supply which charges my old phone in like 10 seconds :)

After doing my own research on the matter -- and it's quite burried unfortunately -- I bought this exact charger from this exact store:

https://www.ravpower.com/products/rp-pc105-61w-pd-wall-charger?_pos=1&_sid=edc3c0ffc&_ss=r

works great with my PPP + keyboard.  As I recall, it doesn't come with a cable; buy a *good* belkin USB-C cable because as I recall part of the problem with the PPP is it requires some of the extra lines on a USB-C cable for power negotiation.  If everything isn't 100% perfect, the PPP will charge at the slowest rate possible.

I really wish the PPP came with a charger given how picky it is :)

tanabi, 2022-05-04:
Also, get one of these: https://pine64.com/product/pinephone-battery-charger/  ... when your PPP's battery inevitably goes entirely flat, this is easiest way to get your PPP back on its feet :)

jsarkari, 2022-05-04:
I can charge my PPP from 0% with a 100W double charger I have, this is when you have towboot install.

jsarkari, 2022-05-04:
Wish I have gone for the one with a usb-a/usb-c instead

tanabi, 2022-05-04:
I'm not 100% sure about that.  I think you can charge the keyboard separately.

One thing to be aware of - do NOT use the phone's USB charge port when it is plugged into the keyboard, because that will damage the keyboard.  The keyboard will probably still work as a keyboard, but it fries the keyboard's ability to charge the phone.  I put electrical tape over the pinephone's charge-hole in the keyboard because I made the mistake once and now I have a parts keyboard :)

tanabi, 2022-05-04:
You could use the cut-out for, say, a USB hub or a USB network adapter or something else that doesn't feed power back into the device.  That's why it's there.  That said, the real puzzle is why they didn't put a couple of diodes on the power line so it doesn't blow up when you plug power into the wrong hole ... or why didn't they put a sticker over the hole with a warning ...

jsarkari, 2022-05-04:
Yeah I tried an usb keyboard or usb network adapter with a usb mouse but it doesn't work - on Arch Linux phosh on SD with Towboot for now.

Danct12, 2022-05-05:
From what I heard, the RK818 PMIC has a safety feature, since the keyboard is feeding power back to the device, it sense that and disable the phone's USB port.

jsarkari01, 2022-05-05:
I havet discovered another issue no matter IF i havet it set to suspend or not when on power and hold the button on keyboard to stop charging the phone so it charges the keyboard battery, after the screen goes off it reverts back to charging the phone instead as it keeps powering up the screen after a few minutes showing the lock screen, no matter what charger I use.

Currently I have managed to get it to charge the keyboard from 0%/to 1% and that's it, it goes back to charging the phone battery.

Once it has charged the phone battery to 100% will it then start charging the keyboard battery?

If not is it software or hardware at fault.

Danct12, 2022-05-05:
Yes, the phone internal battery is charged first, when it's full, it'll start charging the keyboard battery. This post from megi talked about this in detail: https://xnux.eu/log/#065

jsarkari01, 2022-05-05:
Ah OK. The PPP project sure does test your patience to the limit :-)

jsarkari, 2022-05-05:
Another issue I have just encountered is that on inspecting the:

cat /sys/class/power_supply/rk818-battery/status (PPP in keyboard): Discharging

So I took the PPP out of the keyboard and put in the same charger and it is now: Charging

So it is fighting a losing battle if it is discharging in the keyboard and much faster to charge outside of the keyboard then place back in at 100% to hopefully get the keyboard to charge.

But I think there are some issues here and it should need to take many days since I received it to try and get the keyboard to charge.

The problem is even at 100% charged if I place back in the keyboard and it is discharging then I will never get to a point to be able to charge the keyboard.

jsarkari, 2022-05-05:
Finally managed to find a way to charge keyboard with the PPP detached using a high wattage charger and a good charging lead as per above recommendation.  Now up to 50%, plus charging the mobile in the keyboard is way too slow much better to remove and charge out of it for now.

Crsarmv7l, 2022-06-09:
@Danct12 might want to leave this one open. Before I figured out the powersupply, I contacted Megi as suggested. 

He mentioned that he knows of one new protection diode between the keyboard and the phone but doesn't have a unit to test the voltage drop. I am running a few tests for him with a modified voltage_min and the already recommended input_current_limit. Might make regular chargers work better and while not Arch specific it could help others.

Crsarmv7l, 2022-06-15:
Did an initial bit of testing with Megi's suggestions and sent them along to him.

I use his normally recommended "echo 1500000 > /sys/class/power_supply/axp20x-usb/input_current_limit" constantly (startup script). His other recommendation was to try "echo 4000000 > /sys/class/power_supply/axp20x-usb/voltage_min"

Initial results suggests that it does help the phone retain charge, but I need to drain the keyboard battery today, and attempt a charge with one of the many chargers I now have.

Seems like the pi 4 charger is the way to go regardless, but this setting may help.

Crsarmv7l, 2023-02-10:
Ended up desoldering the offending resistor/diode and bridging. The thing is REALLY small, so if you attempt this be very careful

closed 2023-11-10 by jsarkari