dreemurrs-embedded/Pine64-Arch #388: [Plasma-Mobile] pinephone plasma-dialer not starting

hannescam, 2022-05-08:
the output log of the command plasmaphonedialer:

`kf.kirigami: Units.devicePixelRatio is deprecated (since 5.86): This returns 1 when using Qt HiDPI scaling.
kf.kirigami: Units.wheelScrollLines is deprecated (since 5.86): Use Qt.styleHints.wheelScrollLines instead
KPEOPLESINK
kf.kirigami: Units.wheelScrollLines is deprecated (since 5.86): Use Qt.styleHints.wheelScrollLines instead
kf.kirigami: Units.devicePixelRatio is deprecated (since 5.86): This returns 1 when using Qt HiDPI scaling.
file:///usr/lib/qt/qml/org/kde/kirigami.2/AbstractApplicationWindow.qml:282:5: QML Binding: Not restoring previous value because restoreMode has not been set.
This behavior is deprecated.
You have to import QtQml 2.15 after any QtQuick imports and set
the restoreMode of the binding to fix this warning.
In Qt < 6.0 the default is Binding.RestoreBinding.
In Qt >= 6.0 the default is Binding.RestoreBindingOrValue.

file:///usr/lib/qt/qml/org/kde/kirigami.2/AbstractApplicationWindow.qml:282:5: QML Binding: Not restoring previous value because restoreMode has not been set.
This behavior is deprecated.
You have to import QtQml 2.15 after any QtQuick imports and set
the restoreMode of the binding to fix this warning.
In Qt < 6.0 the default is Binding.RestoreBinding.
In Qt >= 6.0 the default is Binding.RestoreBindingOrValue.

Speicherzugriffsfehler (Speicherabzug geschrieben)
`

Danct12, 2022-05-15:
I'm unable to reproduce this crash on a fully updated system.

hannescam, 2022-05-16:
i have fully updated my pinephone 3gb 32gb and still the same error

oihatto64, 2022-05-21:
for me with pinephone I have similar crash . the problem appeared when I switched the language and format in French: with American English  I have no problem

Danct12, 2022-05-21:
Hmm, make sense why I don't see this error.

hannescam, 2022-05-21:
does there exist a fix or should i simply set the language to english