dreemurrs-embedded/Pine64-Arch #390: archlinux-pinephone-phosh-20220502.img.xz  cannot find WiFi in PinePhone BETA

wcdeich4, 2022-05-17:
archlinux-pinephone-phosh-20220502.img.xz  cannot find WiFi in PinePhone BETA

wcdeich4, 2022-05-25:
I bought the beta edition that does not include the dock :(  I guess I could buy a dock.  Is there any way to connect to lan with a dock?

hannescam, 2022-05-30:
no without a dock your have no options to connect to the internet without wifi ecxept you have some sort of usb c to usb a female adaptor and a usb ethernet dongle or a usb c ethernet dongle it is also possible with mobile data

Danct12, 2022-05-31:
I was unable to reproduce this problem, your battery might be too discharged. What happens if you unplug the USB cable?

wcdeich4, 2022-06-03:
Hi, I had a few minutes to test it.  I downloaded a fresh copy of archlinux-pinephone-phosh-20220502.img.xz, put it on a new SD card & tested it w/out the PinePhone BETA without the power cord connected & WiFi worked.  Not sure if I had a bad download, bad SD card or not connecting the usb cable made the difference, but it is working now.  Sorry to bother you.

closed 2022-06-03 by wcdeich4