dreemurrs-embedded/Pine64-Arch #391: [Feature Removal] Remove uboot from pinephone image

luigi311, 2022-05-17:
- Device: (PinePhone or PineTab?) Pinephone
- Kernel Version : $(uname -r) N/A
- UI: N/A

Perhaps we should follow the mobian route of removing the bootloader from the image all together so users can flash tow-boot to the emmc on the pinephone and not have it be overwritten every time they need to flash archlinux again. This will allow the users to keep towboot on the pinephone so we can chose to boot from the emmc or the sd card should other OS follow along.

Danct12, 2022-05-18:
We've been thinking of doing that, the issue now is that U-Boot PinePhone (OG) package also includes PineTab, so we need a PineTab build for Tow-Boot, which some people in the community are looking to do it.

So until then, U-Boot support will remains. But the good news is that you can remove it with a couple of dd commands.

luigi311, 2022-05-18:
is there a dd command you can use on the img? Is it literally the same as if you are doing it to a disk?

sudo dd if=/dev/zero of=archlinux-pinephone-sxmo-20220502.img bs=8k seek=1 count=4

and then flashing that to the pinephone?

shulamy1, 2022-08-10:
Danct12

i tried your dd commands  but the phone doesn't boot from the sd card (the boot procces moove to the sd but nothing happens
ezik

shulamy1, 2022-09-06:
is it hard to make an image without boot loader for the pinephone and leave u-boot for pinetab?
 
ezik

shulamy1, 2023-01-04:
ArenM did it  but you ignore his PR
ezik