dreemurrs-embedded/Pine64-Arch #394: No incoming calls audio, nor ringtone audio on the PinePhone 1.2 since about 2-3 weeks ago

marek-lach, 2022-05-19:
- Device: (PinePhone 1.2 3GB)
- Kernel Version : 5.17.5)
- UI: phosh

### Steps to reproduce
This is difficult. There's no audio for an incoming phone call at all, and no ringtone of an incoming phone call. Works with Manjaro, so this has to be an Arch package specific issue here, not sure which.

### Expected behavior
The phone rings with an incoming call, and I could hear the caller.

### Actual behavior
The phone doesn't ring. It only displays an incoming call UI, but there's no sound of it at all.
I also tried reflashing a completely new image of Arch Linux ARM from scratch with the same problem.
Manjaro currently doesn't have this problem and works, so it has to be some issue with a package related to audio here?

Danct12, 2022-05-19:
How long ago you tried out the new image? If it's a week ago then please read:

https://archlinux.org/news/undone-replacement-of-pipewire-media-session-with-wireplumber/

On Thursday, May 19, 2022 5:37:06 PM +07 Marek Ľach wrote:
> - Device: (PinePhone 1.2 3GB)
> - Kernel Version : 5.17.5)
> - UI: phosh
>
> ### Steps to reproduce
> This is difficult. There's no audio for an incoming phone call at all, and
> no ringtone of an incoming [...]

marek-lach, 2022-05-19:
> How long ago you tried out the new image? If it's a week ago then please read: https://archlinux.org/news/undone-replacement-of-pipewire-media-session-with-wireplumber/
> […](#)
> On Thursday, May 19, [...]

It was just yesterday. I don't think wireplumber has any impact on it, sadly...

philmmanjaro, 2022-05-28:
wireplumber broke the audio stack. We are back to pulseaudio on Manjaro ...

Danct12, 2022-06-01:
I'll be closing this for now, if this is still a issue, please let me know.

closed 2022-06-01 by Danct12