dreemurrs-embedded/Pine64-Arch #402: Installation (images) with F2FS

SoongVilda, 2022-06-01:
I checked what's filesystem use PinePhone and I found PinePhone use EXT4. Why not F2FS? 
F2FS is faster and more responsive on eMMC, SSD or flash memory - performance of memory would be better.

More fixes and improvements going to appear in following (new) kernels 
https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.19-F2FS

Danct12, 2022-06-01:
We already did, we killed a lot of SD cards. I'll explore this option again in the future.

SoongVilda, 2022-06-02:
Maybe we should wait to linux 5.19? https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.19-F2FS

oo-san, 2022-06-11:
I really don't recommend using F2FS. From what I've seen, f2fs is faster but fsck.f2fs is not as robust as fsck.ext4. Anecdotally, I used it for a couple of months on my PP encouraged by [these benchmarks](https://wiki.debian.org/PinePhone/Benchmarks). It ended after I suffered enough data loss on a hard power-loss event that I had to recreate my system from scratch. F2FS weak fsck is even mentioned at the top of [its page on the arch wiki](https://wiki.archlinux.org/title/F2FS#fsck_failures). This is a known issue.

Maybe F2FS is good for people who have a solid backup/snapshotting system in place. Those who are motivated enough to ensure that are also motivated enough to set up F2FS on their own. But I think it's a really bad idea to ship images with F2FS because it's only a matter of time before their systems will fail or become unstable due to file loss and corruption after enough ungraceful power loss events. It really didn't take long for me.

I suggest this be closed as a wontfix or moved to discussions to start a discussion on how to set up F2FS in archlinuxarm.

See also: https://forum.pine64.org/showthread.php?tid=12170

closed 2022-06-21 by Danct12