dreemurrs-embedded/Pine64-Arch #403: Xonotic game in repositories

SoongVilda, 2022-06-02:
It would be possible to add Xonotic game into the repositories? It works as flatpak and I think that's kills a lot performance. 
Current state of gaming experience: Game is enough smooth to target enemies (NPC) on map.

Tested on PinePhone 1.2b (Quad-Core Allwinner A64).
Regards to much more powerful hardware on PinePhone Pro, it would be really smooth on PinePhone Pro (Next reason to have Xonotic natively in repositories and not only flatpak).

Xonotic game website: https://xonotic.org/
Xonotic game GitLab: https://gitlab.com/xonotic
Xonotic PKGBUILDs for ArchLinux x86-64: 
https://github.com/archlinux/svntogit-community/blob/packages/xonotic-data/trunk/PKGBUILD 
https://github.com/archlinux/svntogit-community/tree/packages/xonotic/trunk

Danct12, 2022-06-13:
We do not maintain any package builds from Arch Linux upstream. Please get in touch with Arch Linux ARM developers.

closed 2022-06-13 by Danct12

BillFleming, 2022-09-10:
FYI You can build it straight on the phone using pkgbuild from AUR.

Danct12, 2022-10-11:
We do not maintain anything outside our danctnix/danctnix-testing repos, I highly recommend reaching out to Arch Linux ARM developers as they sync packages from upstream and build them on ARM.