dreemurrs-embedded/Pine64-Arch #407: Could not parse config file /etc/osk.conf

Dan-Sun, 2022-06-04:
- Device: PinePhone and PinePhone Pro
- Kernel Version : 5.17.6-1-danctnix
- UI:  Plasma and phosh (both fde)

### Steps to reproduce
Probably only with fde images. It happened after the updates today. PPP Phosh without fde doesn't have this problem.

### Expected behavior
Boot normal.

### Actual behavior
Before I can add my Luks password, an error message appears.
```
Could not parse config file /etc/osk.conf
Syntax error on line 4
```

Weird is, it looks fine to me.
```
1. # Config file for osk-sdl
2. # The space around the = is important
3. 
4. wallpaper = #000000
5. 
6. keyboard-background = #0E0E12
7. keyboard-map = us
8. keyboard-font = /usr/share/fonts/TTF/DejaVuSans.ttf
9. keyboard-font-size = 24
10. 
11. key-foreground = #FFFFFF
12. key-foreground-highlighted = #0E0E12
13. key-background-letter = #5A606A
14. key-background-return = #003C00
15. key-background-other = #32363E
16. key-background-highlighted = #FFFFFF
17. key-radius = 0
18. key-vibrate-duration = 100
19. key-preview-popup = true
20. 
21. inputbox-foreground = #FFFFFF
22. inputbox-foreground-error = #7E141A
23. inputbox-background = #32363E
24. inputbox-background-error = #E76A63
25. inputbox-radius = 0
26. 
27. animations = true
```

### Logfiles and additional information
It doesn't boot.

Danct12, 2022-06-04:
Fixed in 1ce0ca39a52000b10ae997fe1a83cf255c0a22cc and currently in testing, you'll need to enable [danctnix-testing] repo, update and regenerate initramfs.

Dan-Sun, 2022-06-04:
I tested it, but it doesn't work, at least not on my end.

Dan-Sun, 2022-06-04:
Ups. Forgot to add the boot partition...

So, it works on both.

Thanks for the quick update!

closed 2022-06-04 by Dan-Sun

Danct12, 2022-06-05:
Have you mounted the /boot partition before mounting the chroot?

Dan-Sun, 2022-06-05:
Nah, that's what I meant. I've forgotten that. So it's all good. Everything is running normal again.

Danct12, 2022-06-05:
Ah, ok. Good to hear.

slackerbob, 2022-06-30:
> Unfortunately, it doesn't work for me. After enabling the danctnix-testing repo and upgrading, my pinephone pro booted into a black screen (no backlight). Downgrading linux-megi (from 5.18.3-1 to 5.17.5-1) solved the issue (used my original pinephone for chroot). Yes, the boot partition was mounted. What am I doing wrong?

I had this exact same problem after installing the updated osk-sdl, a black screen instead of the boot messages/splashscreen.

I can confirm that downgrading the kernel fixed this issue and the phone boots again.

I didn't have to enable [danctnix-testing], as the osk-sdl fix is now in the standard repo (and enabling testing didn't give me any more packages).

Pinephone Pro, FDE on mmc

Dan-Sun, 2022-06-30:
@slackerbob 
It is probably more this issue #418 now what you are facing atm.

slackerbob, 2022-07-02:
> @slackerbob It is probably more this issue #418 now what you are facing atm.

Yes, I'd say that's definitely the new issue. Thanks!