dreemurrs-embedded/Pine64-Arch #408: Choosing a kernel to boot from

oz123, 2022-06-05:
- Device: PinePhone
- Kernel Version : │5.12.14-1-danctnix
- UI: Phosh

This is more of a question than a bug, sorry... 

I have the pine phone keyboard  and would like to use the megi-kernel. However, I don't know how to boot into it (I installed the package).

Thanks you for putting up this OS for this phone.

Danct12, 2022-06-06:
You're currently using the Megi kernel, there's no need to do anything right out of the box.

If the keyboard doesn't work, then you need to reseat the keyboard connection.

oz123, 2022-06-06:
Thank you for the quick response.
I read in some issues of waydroid that this module is required, but I do not see it:
```
[alarm@danctnix ~]$ zcat /proc/config.gz | grep PSI
# CONFIG_PSI is not set
```

The config for a newer kernel is already with it:
https://github.com/dreemurrs-embedded/Pine64-Arch/blob/6eef0a70a12a97a63ce3478ddf375c90858019fd/PKGBUILDS/pine64/linux-megi-rc/config#L125

However, this is not found when I do:
```
pacman -Syuu
```

although the PKGBUILD is 3 months old.

Further:
```
[alarm@danctnix ~]$ pacman -Q | grep linux
archlinux-appstream-data 20220512-1
archlinuxarm-keyring 20140119-1
linux-api-headers 5.17.5-2
linux-megi 5.17.6-1
util-linux 2.38-1
util-linux-libs 2.38-1
[alarm@danctnix ~]$ uname -r
5.12.14-1-danctnix
```

I don't understand why `uname -r` is reporting another kernel then `pacman`. Do you have any idea?

Danct12, 2022-06-07:
Hmm, for some reason the kernel seems to have failed to update, are you on FDE setup? Maybe /boot isn't mounted for some reason.

oz123, 2022-06-07:
Yes, I am on FDE. So there is a kernel in /boot partition and another /boot on the root partition?
Didn't think about this possibility...
Any idea how do I mount this?

closed 2023-04-12 by Danct12