dreemurrs-embedded/Pine64-Arch #409: Regression: display autorotation not working after updating from iio-sensor-proxy-3.0-1 to iio-sensor-proxy-3.3-1

dariox86, 2022-06-06:
- Device: PinePhone
- Kernel Version : 5.17.6-1-danctnix
- UI: Phosh

### Steps to reproduce
Upgrade from iio-sensor-proxy-3.0-1 to iio-sensor-proxy-3.3-1, reboot the device, enable display autorotation and tilt the screen.

### Expected behavior
The display is expected to autorotate.

### Actual behavior
The display does not autorotate despite autorotation being enabled.

### Logfiles and additional information
After reverting to iio-sensor-proxy-3.0-1 and rebooting the device, display autorotation works. Maybe issue #191 is related.

Crsarmv7l, 2022-06-07:
Yep, I mentioned this as an afterthought in #405 I hadn't narrowed down the specific package though

Danct12, 2022-06-13:
I'm unable to reproduce this issue on my PinePhone 1.2. Can one of you send your logs? (journalctl -b)

Crsarmv7l, 2022-06-13:
Sure, here is the log.

[pplog.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8887742/pplog.txt)

Danct12, 2022-06-13:
I have found the cause, sadly that right now FreeDesktop GitLab (where iio-sensor-proxy source is hosted) is down due to drive crashes.

So I'll have to wait until they got their data recovered and running again. Finger crossed!

https://www.phoronix.com/scan.php?page=news_item&px=FreeDesktop-GitLab-2022-Crash

Crsarmv7l, 2022-06-13:
Awesome! What do you think the issue is, if you don't mind writing a few lines?

(Also I am going to delete the log, I didn't realize at the time but some network info is in there)

Danct12, 2022-06-13:
This cryptic message:

```
Jun 13 12:18:37 danctnix iio-sensor-prox[2969]: Could not find trigger name associated with /sys/devices/platform/soc/1c2b000.i2c/i2c-1/1-0068/iio:device2
```

Danct12, 2022-06-13:
I have deleted your last comment as editing a comment does not simply remove a log.

Crsarmv7l, 2022-06-13:
Cool thanks for looking into it.

dariox86, 2022-06-13:
FreeDesktop.org [is back online](https://www.phoronix.com/scan.php?page=news_item&px=FreeDesktop-GitLab-Back-Online).

Danct12, 2022-06-13:
It seems like the error you receive is normal: https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/issues/310

Can you edit /usr/lib/systemd/system/iio-sensor-proxy.service? Just uncomment the environment variable thing which should be for debug purposes then reboot the phone and send "journalctl -b 0 -u iio-sensor-proxy.service" output.

dariox86, 2022-06-13:
Here is mine
[journalctl -b 0 -u iio-sensor-proxy.service.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8894548/journalctl.-b.0.-u.iio-sensor-proxy.service.txt)
.

Pinienzapfen, 2022-10-07:
Bumping iio-sensor-proxy to 3.4 fixes display autorotation for me.

dariox86, 2022-10-07:
It does not work for me on version 3.4-1.

Pinienzapfen, 2022-10-08:
Does restarting iio-sensor-proxy fix it for you?

For me it is sporadically broken or working after a reboot.

The cause is a race condition between udev and iio-sensor-proxy:
If udev finishes enumerating the MPU-6050 before iio-sensor-proxy initializes, the sensor works.
If not, the iio-sensor-proxy tries to add the sensor as hotplugged, which logs this debug message:
> Found hotplugged device /sys/devices/platform/soc/1c2b000.i2c/i2c-1/1-0068/iio:device3 of type accelerometer at IIO Poll accelerometer

Hotplugged sensors do not work, the devices are not inserted into the DEVICE_FOR_TYPE map.

Starting/stopping `monitor-sensor --accel` then logs the error message
> driver_set_polling: assertion 'sensor_device' failed

and the accelerometer does not work.

I'll submit the fix upstream when I get access to the freedesktop GitLab:

```
diff --git a/src/iio-sensor-proxy.c b/src/iio-sensor-proxy.c
index 9833be2..1eedf29 100644
--- a/src/iio-sensor-proxy.c
+++ b/src/iio-sensor-proxy.c
@@ -955,11 +955,14 @@ sensor_changes (GUdevClient *client,
 					 driver_type_to_str (driver->type),
 					 driver->driver_name);
 
-				if (driver_open (driver, device,
-						 driver_type_to_callback_func (driver->type), data)) {
+				SensorDevice *sensor_device = driver_open (driver, device,
+					driver_type_to_callback_func (driver->type), data);
+
+				if (sensor_device) {
 					GHashTable *ht;
 
 					UDEV_DEVICE_FOR_TYPE(driver->type) = g_object_ref (device);
+					DEVICE_FOR_TYPE(driver->type) = sensor_device;
 					DRIVER_FOR_TYPE(driver->type) = (SensorDriver *) driver;
 					send_driver_changed_dbus_event (data, driver->type);

```

Enabling debug logging makes the issue less likely to appar since it delays the initialization, so udev gets more time to do its thing.

dariox86, 2022-10-08:
I tried one more time and it worked out of the box.

Pinienzapfen, 2022-10-08:
Submitted [upstream](https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/merge_requests/362).

Danct12, 2022-10-11:
So the daemon does indeed detect a accelerometer, but for some reason it failed to claim it.

```
giu 13 23:15:27 danctnix iio-sensor-prox[2941]: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
giu 13 23:15:27 danctnix iio-sensor-prox[2941]: driver_set_polling: assertion 'sensor_device' failed
```

Maybe you can go report this to upstream and see what they think about it? https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/issues

shulamy1, 2023-01-04:
i am in 3.4.1 now and autorotation doesn't work
ezik

shulamy1, 2023-01-04:
beta edition  arch updated
ezik

dariox86, 2023-01-04:
I am on 3.4.1 and it is not working for me either.

Pinienzapfen, 2023-01-12:
The [fix](https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/merge_requests/362) was merged into master, so it will be resolved in 3.5.

shulamy1, 2023-01-26:
3.4.1 works on manjaro

ezik

Jeremiah-Roise, 2023-04-13:
For anybody who wants a temporary fix until iio-sensor-proxy v3.5 is released just delay `iio-sensor-proxy` by adding `ExecStartPre= /bin/sleep 5` to
`/usr/lib/systemd/system/iio-sensor-proxy.service`

Danct12, 2023-07-27:
iio-sensor-proxy 3.5 just released two days ago.

Danct12, 2024-03-09:
This has been fixed a while ago and v3.5 is in repo.

closed 2024-03-09 by Danct12