dreemurrs-embedded/Pine64-Arch #413: Regression: screen turning on on its own after updating from phoc-0.13.1-1 to phoc-0.20.0-1

dariox86, 2022-06-09:
- Device: PinePhone
- Kernel Version : 5.18.0-2-danctnix
- UI: Phosh

### Steps to reproduce
Upgrade from phoc-0.13.1-1 to phoc-0.20.0-1 and reboot. When the display is on push the power button to turn it off.

### Expected behavior
The screen should stay off.

### Actual behavior
Shortly after the screen was turned off, the display turns on on its own, it shows the boot splash for a few seconds and returns to the lock screen.

### Logfiles and additional information
Reverting to phoc-0.13.1-1 fixes the issue.

dariox86, 2022-06-09:
Not sure either. It looks like it is rebooting, but it is too fast. [Here](https://files.fm/u/8d8hpdsre) is the package from my cache. Just open a terminal emulator from the same directory of the downloaded package and issue this command:
`# pacman -U phoc-0.13.1-1-aarch64.pkg.tar.xz`
Of course, this is not the recommended route from a security perspective. You should have downloaded the relevant PKGBUILD of the previous version of the package and build it with makepkg.

Danct12, 2022-06-10:
Please try again with phoc from danctnix-testing repo, this should now be fixed.

dariox86, 2022-06-10:
It does work for me. Thank you.

Danct12, 2022-06-11:
I've pushed a fixed version to main repo.

closed 2022-06-11 by Danct12