dreemurrs-embedded/Pine64-Arch #421: imx258 doesn't have a devnode

kgmt0, 2022-06-25:
- Device: PinePhone Pro
- Kernel Version : 5.18.4
- UI: Phosh

### Steps to reproduce

1. Run `media-ctl -d 0 -p`.
2. See if you can find 'imx258 1-001a' in the output. If not, repeat step 1, increasing the argument of `-d` by 1.
3. Take note of whether or not there's a devnode by checking if there's a line starting with `device node name` in the output for 'imx258 1-001a'.

### Expected behavior

imx258 has a devnode.

### Actual behavior

imx258 doesn't have a devnode.

### Logfiles and additional information

I'm able to get imx258 to have a devnode by modifying the PKGBUILD to use config from arch/arm64/configs/pinephone_pro_defconfig in the kernel directory. To allow /boot to be mounted, I also enabled CONFIG_NLS_ASCII.

Here are the logs showing the outputs from `media-ctl` before and after switching the kernel:
[before.log](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8985255/before.log)
[after.log](https://github.com/dreemurrs-embedded/Pine64-Arch/files/8985257/after.log)

Danct12, 2022-06-27:
Ah, so this is a missing kernel config problem. I'll look into this later, thanks.

bakerk98, 2022-07-20:
This may be related to the camera not working yet on PPP https://wiki.pine64.org/wiki/PinePhone_Pro/IMX258_Camera_Debugging

kgmt0, 2022-07-27:
This seems to be fixed as of version 5.18.7.

closed 2022-07-27 by kgmt0