dreemurrs-embedded/Pine64-Arch #423: Pinephone pro extremely fast media playback

caveman250, 2022-07-05:
- Device: PinePhone Pro
- Kernel Version : 5.18.4-1.danctnix
- UI: Phosh

### Steps to reproduce
I am unsure of how to reproduce the issue,. but intermittently my pinephone pro will playback any type of media several times faster than it should be. Behavior can be seen across multiple apps including Firefox, Spot, Headlines, Amberol ect. I also have no idea how to fix the issue, it just goes away after a day or two. rebooting does not seem to make a difference.

### Expected behavior
Media plays back at the correct speed.

### Actual behavior
Media plays back very fast.

[journalctl.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/9050282/journalctl.txt)

caveman250, 2022-07-05:
Just realized my journalctl contained multiple session sorry about that. The issue was occuring in the most recent session when i took that log.

Danct12, 2022-07-06:
Duplicate of #381

closed 2022-07-06 by Danct12