dreemurrs-embedded/Pine64-Arch #425: android-tools broken - libprotobuf mismatch

aroggeband, 2022-07-22:
- Device: Pinephone
- Kernel Version : 5.18.7-1 danctnix
- UI: phosh
- android-tools 31.0.3-3
- protobuf 21.2-1
### Steps to reproduce

enter the command:
adb --help

### Expected behavior

adb returns usage text

### Actual behavior

adb: error while loading shared libraries: libprotobuf.so.30: cannot open shared object file: No such file or directory

### Logfiles and additional information

This has been reported and fixed in other distros,
e.g. https://forum.artixlinux.org/index.php/topic,4046.0.html
I am currently dealing with a bricked modem due to a crash in the flash script, possibly related to this issue.

Danct12, 2022-07-22:
This is a upstream issue, I suggest asking about it in #archlinuxarm on Libera.

hamblingreen1, 2022-09-01:
cant reproduce with adb 1.0.41

closed 2022-10-12 by Danct12