dreemurrs-embedded/Pine64-Arch #437: on-screen keyboard

shulamy1, 2022-08-17:
the top line of the keyboard of the terminal (console) disappears 
i'm not sure why but i think by tapping just above it

- Device:  pinephone beta edition
- Kernel Version :  5.9.1-1-danctnix
- UI:  phosh

### Logfiles and additional information
which log to post ?

ezik

mikeshenson, 2022-08-17:
Squeekboard has different keyboard layouts depending on what input is expected. The layout should change when you go into a terminal. The different layouts are default, email, emoji, number, terminal and url.
see:
https://gitlab.gnome.org/World/Phosh/squeekboard/-/tree/master/data/keyboards

I personally dislike keyboard layouts changing so I created my own.
https://github.com/mikeshenson/Phosh-Squeekboard-Keyboard

shulamy1, 2022-08-25:
i forgot to write that the line is there when i open terminal (the line with "ctrl" and "alt" and the arrows)

but after some time  (i think when i miss tap the exect point and tap above the line) the line disappear

ezik

shulamy1, 2022-10-03:
i found a solution somewere else 

choose "terminal" keyboard

ezik

closed 2022-10-03 by shulamy1