dreemurrs-embedded/Pine64-Arch #446: location on maps

shulamy1, 2022-09-01:
- Device: PinePhone
- Kernel Version : 5.19.3-1-danctnix (and older)
- UI: phosh

### Steps to reproduce
open maps and tap  "my location"

### Expected behavior
curser on my location

### Actual behavior
curser about 100km south from my location

Ferenc-, 2022-09-02:
That is most likely because your location is not determined from your GNSS coordinates,
but by other means from Mozilla Location Services.
Check with [satellite](https://flathub.org/apps/details/page.codeberg.tpikonen.satellite), that you even have a GPS fix, and you have exact latitude and longitude values.

shulamy1, 2022-09-04:
yes. you are right
now i have to find how to do it
ezik

closed 2022-09-04 by shulamy1