dreemurrs-embedded/Pine64-Arch #448: Mobile Data on Pinephone w/ Phosh?

AblatedSprocket, 2022-09-07:
Is there some setup required to get mobile data working? I've only read that it works out of the box, but I'm not having that experience. Everything else seems to work okay except MMS, but I'm on Verizon for now so I don't expect that to work.

mikeshenson, 2022-09-07:
I had issues getting a Verizon (USA) sim to work. 

I know you need to have the same APN for internet and MMS.
See Chatty issue:
https://source.puri.sm/Librem5/chatty/-/issues/30#note_157781

I use T- Mobile (USA). Here is the T-Mobile APN setup
https://wiki.pine64.org/wiki/PinePhone_APN_Settings#Patriot_Mobile_(USA_T-Mobile)

Also use Carrier support as a reference
https://wiki.pine64.org/wiki/PinePhone_Carrier_Support

AblatedSprocket, 2022-10-04:
Oh, thanks for the mention @langfingaz. For the record, this never got resolved, but I switched carriers so it's no longer an issue.

closed 2022-10-04 by AblatedSprocket