dreemurrs-embedded/Pine64-Arch #462: build megi's PinePhone kernel with the CONFIG_KEYBOARD_PINEPHONE module

marek-lach, 2022-09-11:
Currently, the PinePhone's physical hardware keyboard is currently not really working now under Arch. Perhaps because of [this](https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-pinephone/-/issues/17)?

- Device: PinePhone
- Kernel Version: 5.19.3
- UI: phosh

### Steps to reproduce
Attach the PinePhone 's physical keyboard, currently it's not really charging/typing.

### Expected behavior
Physical keyboard works.

Danct12, 2022-10-12:
Please give me your dmesg output.

marek-lach, 2022-10-14:
> Please give me your dmesg output.

[    0.000000] Machine model: Pine64 PinePhone (1.2)
[    0.000000] earlycon: uart0 at MMIO32 0x0000000001c28000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xff9b4040-0xff9b5fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] cma: Reserved 256 MiB at 0x00000000ec800000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 20 pages/cpu s43112 r8192 d30616 u81920
[    0.000000] pcpu-alloc: s43112 r8192 d30616 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 774144
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: loglevel=4 console=tty0 console=ttyS0,115200 earlycon=uart,mmio32,0x01c28000 consoleblank=0 boot=PARTUUID=af3af8c6-01 root=PARTUUID=af3af8c6-02 rw rootwait quiet audit=0 bootsplash.bootfile=bootsplash-themes/manjaro/bootsplash
[    0.000000] Unknown kernel command line parameters "boot=PARTUUID=af3af8c6-01 audit=0", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 2793712K/3145728K available (12224K kernel code, 1254K rwdata, 3568K rodata, 2816K init, 462K bss, 89872K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000] Trampoline variant of Tasks RCU enabled.
[    0.000000] Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: Enabling global workaround for Allwinner erratum UNKNOWN1
[    0.000000] arch_timer: CPU0: Trapping CNTVCT access
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000244] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000946] Console: colour dummy device 80x25
[    0.001011] printk: console [tty0] enabled
[    0.001108] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.001126] pid_max: default: 32768 minimum: 301
[    0.001200] LSM: Security Framework initializing
[    0.001349] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001388] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.003498] cblist_init_generic: Setting adjustable number of callback queues.
[    0.003519] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.003627] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.003917] rcu: Hierarchical SRCU implementation.
[    0.003925] rcu: Max phase no-delay instances is 1000.
[    0.005667] smp: Bringing up secondary CPUs ...
[    0.007419] Detected VIPT I-cache on CPU1
[    0.007570] arch_timer: CPU1: Trapping CNTVCT access
[    0.007590] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[... 500 lines cut ...]
[  276.728734] power_supply ip5xxx-battery: driver failed to report `constant_charge_current' property: -5
[  276.741072] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[  276.751983] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[  276.762813] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[  276.766330] power_supply ip5xxx-battery: driver failed to report `current_now' property: -5
[  276.780647] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[  276.791493] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[  276.807128] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[... cut ...]
[  310.546329] power_supply ip5xxx-battery: driver failed to report `status' property: -5
[  310.557405] power_supply ip5xxx-battery: driver failed to report `current_now' property: -5

From the wiki, this part seems to be relevant:

Kernel-space driver

Kernel driver implementation from Samuel Holland: CONFIG_IP5XXX_POWER (module ip5xxx_power) and CONFIG_KEYBOARD_PINEPHONE (module pinephone_keyboard) https://github.com/smaeul/linux/commits/wip/pp-keyboard

Note: If you've upgraded to >=5.18, don't forget to upgrade the dtb as kb151 now appears to be a stub.

Danct12, 2022-10-17:
To me it sounds like you need to reseat the keyboard module with a shim.

bakerk98, 2022-10-25:
> To me it sounds like you need to reseat the keyboard module with a shim.

OP, you can also get the replacement POGO pins from the Pine64 store and put them in your Pro, it helps it contact the kb better, and won't eventually bulge the back of the kb

Danct12, 2023-08-29:
Since the kernel is compiled with the module, it's possible that your keyboard isn't making good contacts with the pogo pins on your device. You can try to reseat the case and see if that fixes it.

closed 2023-08-29 by Danct12