dreemurrs-embedded/Pine64-Arch #469: How to find what devices its compatable with?

Xanthus58, 2022-10-05:
Hey so I'm sorry for posting this as a bug report but i don't see a discussion section here on git-hub. (Which is where i found this project) 

Now I've been using arch Linux on my desktop for over a year now and i love it. So i was trying to find a way to move my mobile device across too (from android) And according to what i can see everyone using a "Pinephone" but i was wondering if there's a way to figure out if its compatible with existing devices? I currently have a samsumg S9 and a Nokia TA-1206 that I'd like to convert if possible.

DrRac27, 2022-10-05:
This project only focuses on Pine64 devices like the name says but you can check out PostmarketOS and the [supported devices list](https://wiki.postmarketos.org/wiki/Devices). In theory you could create some PKGBUILDs for your phones and use the Arch ARM and Danctnix repos for the rest if your phones are aarch64 but it is not possible right now to run this distro on your phones.

Xanthus58, 2022-10-05:
Thanks a ton this is exactly what i was looking for i appreciate you taking the time to reach out and answer my question. I'll go ahead and look into the various things you mentioned and possibly getting a pinephone at some point.

closed 2022-10-05 by Xanthus58