dreemurrs-embedded/Pine64-Arch #486: Support for the Pinebook Pro

UltraBlackLinux, 2022-10-23:
Hey there,
I would be pretty happy if this project was to support the Pinebook Pro. Currently there are no prebuilt arch images available for it, resulting in Manjaro being the only arch image available.

I don't like manjaro and would like to use something proper.

Could you start supporting it? The name of this repo implies that it already does ;) 
Thanks!

Danct12, 2022-10-24:
https://github.com/SvenKiljan/archlinuxarm-pbp

closed 2022-10-24 by Danct12

UltraBlackLinux, 2022-10-24:
I have already seen this, but sadly this does not appear to be producing flashable images like the ones from danctnix.

I don't have the experience to set up arch myself, and even with the provided instructions, I couldn't even create a flashable image just because I have no clue how, without being left with a 150G image.

I need an image, because I want to be able to reflash it at any point if I need to.

Danct12, 2022-10-25:
Upstream Arch Linux ARM doesn't include flashable images either.

Honestly, I do not want to support something that someone have already done. But this is something you can try to talk to the maintainer, and if they don't want it then I guess just try and set it up manually.

I mean, the installation guide doesn't look THAT complicated and somewhat close to the upstream Arch Linux ARM's guide. Just try to set it up yourself, you may learn something.