dreemurrs-embedded/Pine64-Arch #489: BTRFS Filesystem

UltraBlackLinux, 2022-10-29:
Hey there,
could you add support for BTRFS? It seems like proper backups are quite important on this kind of beta platform.
Thanks!

DrRac27, 2022-10-29:
You can already use btrfs (I do). There are multiple options, for example:
1. Boot from SD card and install manually on internal storage
2. Use the squashfs files to install it: https://danctnix.arikawa-hi.me/rootfs/archarm-on-mobile/
3. Use the [FDE script](https://github.com/dreemurrs-embedded/archarm-mobile-fde-installer) and modify it for your needs. This is how I did it. If you want I can create a PR?

UltraBlackLinux, 2022-10-29:
> You can already use btrfs (I do). There are multiple options, for example:
>
>     1. Boot from SD card and install manually on internal storage
>
>     2. Use the squashfs files to install it: [...]

2. Does that have BTRFS images? I also need FDE
3. This does not  officially support the PPP yet. I'm already using the fork that does, and even still I'm not really experienced in creating my own btrfs filesystem.

DrRac27, 2022-10-29:
I created the PR: https://github.com/dreemurrs-embedded/archarm-mobile-fde-installer/pull/22
I also created a branch where I merged the PR for PPP support for you: https://github.com/DrRac27/archarm-mobile-fde-installer/tree/PPP-btrfs
Have fun!

UltraBlackLinux, 2022-10-29:
> I created the PR: [dreemurrs-embedded/archarm-mobile-fde-installer#22](https://github.com/dreemurrs-embedded/archarm-mobile-fde-installer/pull/22) I also created a branch where I merged the PR for PPP support for you: https://github.com/DrRac27/archarm-mobile-fde-installer/tree/PPP-btrfs (not tested as I have no PPP) Have fun!

oh thank you! <3

closed 2022-10-29 by UltraBlackLinux