dreemurrs-embedded/Pine64-Arch #490: Phosh, changed scaling: Screen stays permanently black after wake up

UltraBlackLinux, 2022-10-29:
- Device: (PinePhone or PineTab?) Pinephone Pro
- Kernel Version : $(uname -r) Linux pinecone 6.0.2-3-danctnix #1 SMP PREEMPT_DYNAMIC Sat Oct 22 03:23:48 UTC 2022 aarch64 GNU/Linux
- UI: Phosh

### Steps to reproduce
Change the default display scaling once, to anything, e.g. via gnome-settings

### Expected behavior
It should not break phosh

### Actual behavior
Any consecutive sleeps of the device will be permanent. You will not be able to wake it up, even after rebooting, clearing the cache, removing ~/.local/share/gnome-shell
Even still the blinking notification led will stop blinking, indicating that it has in fact woken up, though the display stays black.
A warning can be seen in the journal when trying to do so: `gsd-color[3851]: unable to get EDID for xrandr-DSI-1: unable to get EDID for output`

I did get it to wake up a 4 times while trying to debug this. The first wake up took about 5 minutes, then the other four more were as slow as waking the PPP up from suspense, and then it stayed dead again.

### Logfiles and additional information
```
Oct 29 14:35:52 pinecone systemd-logind[3400]: Power key pressed short.
Oct 29 14:35:52 pinecone kernel: panel-himax-hx8394 ff960000.mipi.0: disable
Oct 29 14:35:52 pinecone kernel: dw-mipi-dsi-rockchip ff960000.mipi: disable
Oct 29 14:35:52 pinecone kernel: dw-mipi-dsi-rockchip ff960000.mipi: post disable
Oct 29 14:35:52 pinecone kernel: dw-mipi-dsi-rockchip ff960000.mipi: mode 0
Oct 29 14:35:52 pinecone kernel: panel-himax-hx8394 ff960000.mipi.0: unprepare
Oct 29 14:35:52 pinecone gsd-color[3931]: unable to get EDID for xrandr-DSI-1: unable to get EDID for output
```

UltraBlackLinux, 2022-10-30:
It turns out, that the backlight is just not turned on by the system.

UltraBlackLinux, 2022-10-30:
The backlight works when manually overriding it via e.g. `echo 1000 > /sys/class/backlight/backlight/brightness`

Touching the brightness slider in gnome breaks it again.

megous, 2022-10-30:
If it's just backlight, then your log doesn't make any sense. Log ends with display panel completely disabled. If you'd just enable the backlight via sysfs at that point you'd see backlit black screen, with everything else still disabled.

UltraBlackLinux, 2022-10-30:
> If it's just backlight, then your log doesn't make any sense. Log ends with display panel completely disabled. If you'd just enable the backlight via sysfs at that point you'd see backlit black screen, with everything else still disabled.

I don't understand. I mean I don't understand anything about this problem really. I just now that it now works

closed 2022-10-30 by UltraBlackLinux