dreemurrs-embedded/Pine64-Arch #491: phosh fails to start, libgnome-desktop-3.so.19 missing

langfingaz, 2022-11-03:
- Device: PinePhone
- Kernel Version : 6.0.2-3-danctnix
- UI: Phosh

### Steps to reproduce

Since the last update (`pacman -Syu`), phosh does no longer start.

### Logfiles and additional information

* Manually starting `phosh` (via ssh):

```
phoc: error while loading shared libraries: libgnome-desktop-3.so.19: cannot open shared object file: No such file or directory
```

* `journalctl -xe`

```
Nov 03 10:33:08 danctnix systemd-logind[3073]: New session 168 of user alarm.
░░ Subject: A new session 168 has been created for user alarm
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ Documentation: sd-login(3)
░░ 
░░ A new session with the ID 168 has been created for the user alarm.
░░ 
░░ The leading process of the session is 5131.
Nov 03 10:33:08 danctnix systemd[1]: Started Session 168 of User alarm.
░░ Subject: A start job for unit session-168.scope has finished successfully
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit session-168.scope has finished successfully.
░░ 
░░ The job identifier is 26043.
Nov 03 10:33:08 danctnix audit[5131]: USER_START pid=5131 uid=0 auid=1000 ses=168 subj=unconfined msg='op=PAM:session_open grantors=pam_loginuid,pam_keyinit,pam_systemd_home,pam_limits,pam_unix,pam_permit,pam_mail,pam_systemd,pam_env acct="alarm" exe="/usr/lib/systemd/systemd" hostname=danctnix addr=? terminal=/dev/tty7 res=success'
Nov 03 10:33:08 danctnix kernel: audit: type=1105 audit(1667467988.694:1402): pid=5131 uid=0 auid=1000 ses=168 subj=unconfined msg='op=PAM:session_open grantors=pam_loginuid,pam_keyinit,pam_systemd_home,pam_limits,pam_unix,pam_permit,pam_mail,pam_systemd,pam_env acct="alarm" exe="/usr/lib/systemd/systemd" hostname=danctnix addr=? terminal=/dev/tty7 res=success'
Nov 03 10:33:08 danctnix phosh[5131]: /usr/bin/phoc: error while loading shared libraries: libgnome-desktop-3.so.19: cannot open shared object file: No such file or directory
Nov 03 10:33:08 danctnix systemd[1]: phosh.service: Main process exited, code=exited, status=127/n/a
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ An ExecStart= process belonging to unit phosh.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 127.
Nov 03 10:33:08 danctnix systemd[1]: phosh.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit phosh.service has entered the 'failed' state with result 'exit-code'.
Nov 03 10:33:08 danctnix audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=unconfined msg='unit=phosh comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Nov 03 10:33:08 danctnix kernel: audit: type=1131 audit(1667467988.726:1403): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=unconfined msg='unit=phosh comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Nov 03 10:33:08 danctnix systemd[1]: session-168.scope: Deactivated successfully.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit session-168.scope has successfully entered the 'dead' state.
Nov 03 10:33:08 danctnix systemd-logind[3073]: Session 168 logged out. Waiting for processes to exit.
Nov 03 10:33:08 danctnix systemd-logind[3073]: Removed session 168.
░░ Subject: Session 168 has been terminated
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ Documentation: sd-login(3)
░░ 
░░ A session with the ID 168 has been terminated.
```
[... 63 lines cut ...]
[2022-11-03T10:06:12+0100] [ALPM] installed libshumate (1.0.2-1)
[2022-11-03T10:06:12+0100] [ALPM] upgraded gnome-maps (42.3-1 -> 43.1-1)
[2022-11-03T10:06:12+0100] [ALPM] installed python-decorator (5.1.1-1)
[2022-11-03T10:06:12+0100] [ALPM] installed python-validators (0.20.0-1)
[2022-11-03T10:06:12+0100] [ALPM] installed python-zxcvbn (4.4.28-2)
[2022-11-03T10:06:12+0100] [ALPM] upgraded gnome-passwordsafe (6.5-2 -> 7.0-1)
[2022-11-03T10:06:13+0100] [ALPM] upgraded gnome-session (42.0-1 -> 43.0-1)
[2022-11-03T10:06:13+0100] [ALPM] upgraded xorg-xwayland (22.1.4-3 -> 22.1.5-1)
[2022-11-03T10:06:14+0100] [ALPM] upgraded mutter (42.5-1 -> 43.0-1)
[2022-11-03T10:06:14+0100] [ALPM] upgraded libnma (1.10.2-1 -> 1.10.2-2)
[2022-11-03T10:06:15+0100] [ALPM] upgraded gnome-shell (1:42.5-1 -> 1:43.0-2)
[2022-11-03T10:06:15+0100] [ALPM] installed endeavour (43.0-1)
[2022-11-03T10:06:15+0100] [ALPM] upgraded liboauth (1:1.0.3+r15+gac4cf3a-1 -> 1:1.0.3+r16+gc26f038-1)
[2022-11-03T10:06:16+0100] [ALPM] upgraded totem-pl-parser (3.26.6-1 -> 3.26.6-2)
[2022-11-03T10:06:16+0100] [ALPM] upgraded grilo (0.3.15-1 -> 0.3.15-2)
[2022-11-03T10:06:16+0100] [ALPM] upgraded gssdp (1.4.0.1-1 -> 1.6.0-1)
[2022-11-03T10:06:16+0100] [ALPM] upgraded gst-plugin-pipewire (1:0.3.59-2 -> 1:0.3.59-3)
[2022-11-03T10:06:16+0100] [ALPM] upgraded gupnp (1:1.4.3-1 -> 1:1.6.0-1)
[2022-11-03T10:06:16+0100] [ALPM] upgraded gupnp-igd (1.2.0-2 -> 1.2.0-3)
[2022-11-03T10:06:16+0100] [ALPM] upgraded mpg123 (1.30.2-1 -> 1.31.1-1)
[2022-11-03T10:06:17+0100] [ALPM] upgraded poppler (22.10.0-1 -> 22.11.0-1)
[2022-11-03T10:06:18+0100] [ALPM] upgraded poppler-glib (22.10.0-1 -> 22.11.0-1)
[2022-11-03T10:06:19+0100] [ALPM] upgraded python-setuptools (1:63.4.3-1 -> 1:64.0.0-1)
[2022-11-03T10:06:19+0100] [ALPM] transaction completed
```

dariox86, 2022-11-03:
I am having the same issue. I tried downgrading everything to no avail. Perhaps I am missing something.

Danct12, 2022-11-03:
Phoc will need a rebuild, I'll look at this later for any other necessary upgrades.

dariox86, 2022-11-03:
It took me hours to downgrade everything and yet there is some conflict I have yet to work out. I am left with an unbootable device.

dariox86, 2022-11-03:
Update: after spending eight hours trying to figure out a workaround, right before giving up I found out I had to downgrade all GNOME related packages from version 43 to version 42.

xethos84, 2022-11-03:
An easier method (read: dirty hack) is to soft-link libgnome-desktop-3.so.20 to libgnome-desktop-3.so.19

    sudo ln -s /usr/lib/libgnome-desktop-3.so.20 /usr/lib/libgnome-desktop-3.so.19

Not *great*, but bootable. I used a uSD card I had with pmOS to chroot into my borked system, link the library, and reboot - not the worst solution without a proper keyboard, if it helps anyone else.

dariox86, 2022-11-03:
I booted from a microSD with Arch Linux ARM DanctNIX and used arch-chroot from the live system.

Danct12, 2022-11-04:
There are many things needs to be rebuilt on the way, right now phoc and phosh is done (and boots to a crash screen).

Danct12, 2022-11-04:
squeekboard is also done, this should now allow phosh to boot all the way.

However, chatty will have to be remain broken as I'm trying to get it to rebuild and failed as evolution-data-server uses libsoup3. In worst cases, I'll have to use a 0.7.0_rcX build (I'm so sorry) or get rid of the experimental matrix implementation.

I'll build gnome-calls 43, so hopefully that should be fixed in the same day.

Danct12, 2022-11-04:
Control Center should now work, as well as many other package as they've been upgraded from 42 to 43.

langfingaz, 2022-11-04:
Many thanks for the quick rebuilds! Just did an update via ssh and the UI is running again.

closed 2022-11-04 by langfingaz