dreemurrs-embedded/Pine64-Arch #492: Adding apps to the new home screen disappear after a reboot

jsarkari, 2022-11-04:
- Device: (PinePhone or PineTab?) PinePhone Pro
- Kernel Version : $(uname -r) 6.0.2-3-dantix
- UI: KDE Plasma 5.26.2

### Steps to reproduce

Adding apps to the new homescreen as favourites. Then after a reboot they have disappeared.

### Expected behavior

After a restart or reboot the favourite apps should still apear on the new homescreen.

### Actual behavior

The favourites apps disappeared on restart.

### Logfiles and additional information

grinceur, 2022-11-04:
yep need to add back the `X-Purism-FormFactor` line in the `.desktop` files

grinceur, 2022-11-04:
> Is it possible to add that myself as a workaround until it is added. If so where is this file? Thanks
> […](#)

hi i was mistaken, this will not solve your issue sorry i misread your report…

closed 2023-11-10 by jsarkari