dreemurrs-embedded/Pine64-Arch #498: Mkimage not working after latest update.

Laar3, 2022-11-19:
- Device: PinePhone
- Kernel Version : $(uname -r)
- UI: 

### Steps to reproduce
Do an update then run try to use mkimage to convert `/boot/boot.txt` to `/boot/boot.scr`.

### Expected behavior
The program to convert the `boot.txt` to `boot.scr`

### Actual behavior
libssl produces the following error
```
mkimage: error while loading shared libraries: libssdl.so.1.1: cannot open shared object file: No such file or directory
```

I believe this is due to an update to libssl and that the `uboot-tools` package needs to be rebuilt, i think thats all thats needed but im no expert.

Danct12, 2022-11-21:
Another user having the same issue: https://archlinuxarm.org/forum/viewtopic.php?f=15&t=16247

Danct12, 2022-12-05:
This should now be fixed upstream.

closed 2022-12-05 by Danct12