dreemurrs-embedded/Pine64-Arch #499: Phone Call Configuration - Cannot be heard on phone calls

Raltrwx, 2022-12-02:
- Device: PinephonePro
- Kernel Version : 6.0.8-1-danctnix
- UI: Phosh

### Steps to reproduce
1. Open Settings App -> Sound
2. Open "Configuration" submenu and change to "Phone Call" configuration.

OR

1. Take a phone call

### Expected behavior
For the "Phone Call" configuration to have a configuration which picks up my voice. 

### Actual behavior

The device will no longer be picking up noise, until you close the settings app and then re-open and set it back to "Default".

To further confirm, I installed alsamixer and watched what happened to the input levels when I changed from "Default" to "Phone Call". It immediately reduces "ADC Input" from 100 (Or whatever you set it to) to 19, and always 19.

### Logfiles and additional information
Attached! 
[journaldump.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/10137350/journaldump.txt)

I didn't know I was having so many MMS issues after reviewing my journal, looks like I'll be looking into that next. Sorry the journal is so filled with MMS noise. I am currently trying to figure out how to edit "Gnome Sound Configurations" for lack of a better term so I can compare "Default" to "Phone Call".

Alaraajavamma, 2023-03-12:
I think I found how to fix this.
This is currently setup in PinePhonePro-VoiceCall.conf

		# Set capture volume to 0dB so modem audio doesn't overdrive the ADC
		cset "name='ADC Capture Volume' 47"
		cset "name='Mono ADC Capture Volume' 47"

Microphone is okay with postmarketOS and these lines are like this:

		# Set capture volume to 0dB so modem audio doesn't overdrive the ADC
		cset "name='ADC Capture Volume' 160"
		cset "name='Mono ADC Capture Volume' 160"

So I think these should be the default setup? 
This works (tested today) - but there is issue that somehow microphone drops lower if I answer call which comes when PPP is suspended. I don't know where it comes from but it is the same with pmos.

ewtoombs, 2023-06-18:
I looked for "ADC Input" in the alsa controls, @Raltrwx , but I couldn't find it. Has it been renamed? I'm running 6.3.5-1-danctnix right now.