dreemurrs-embedded/Pine64-Arch #503: Back camera is not working since kernel 6.x

pinoverclock, 2022-12-15:
- Device: PinePhone
- Kernel Version: 6.0.10-1-aarch64
- UI: Phosh

Back camera is not working since kernel 6.x

Danct12, 2023-01-19:
Please check out the kernel in testing repo and see if this fixes the issue for you.

pinoverclock, 2023-01-22:
I can confirm the issue is still present in kernel 6.1.7-1

Danct12, 2023-01-22:
That is weird, mine is working (v1.2b). Do you have any camera turned off?

pinoverclock, 2023-01-24:
Yes, same pinephone version, all switches in the on position. 

I decided to open the Megapixels app while follwing journald log and I found something funny
When I start the Megapixels I get:
```
Jan 24 18:54:44 pp64 kernel: ov5640 4-004c: ov5640_set_ctrl_focus: can't set focus when not powered
Jan 24 18:54:47 pp64 phosh[3465]: Startup of app 'Megapixels' with startup id: '8da32141-6f8e-41d9-8bbc-a0577e082c46' timed out
```
And I have Megapixels running but cycling from black to white patterns, but once I close it I get:
```
Jan 24 18:54:55 pp64 org.postmarketos.Megapixels.desktop[10962]: Found config file at /etc/megapixels/config/pine64,pinephone-1.2.ini
Jan 24 18:54:55 pp64 org.postmarketos.Megapixels.desktop[10962]: Adding camera rear from config
Jan 24 18:54:55 pp64 org.postmarketos.Megapixels.desktop[10962]: Adding camera front from config
Jan 24 18:54:55 pp64 org.postmarketos.Megapixels.desktop[10962]: Initialized OpenGL ES 2.0
```

I tryed it a few times with the same results
[VideoDemo link valid for 24 hours](https://www.veed.io/view/1f773088-9f5e-433d-8e11-adb658473a90?sharingWidget=true&panel=share)

Pigpog, 2023-02-26:
I have yet to see Megapixels support 6.X (maybe im just out of touch).
Instead, I use the program `qcam` from the `libcamera-tools` package, which works very well (although it's not a great UX).
One thing to note is that if the phone ever suspends after boot up, the cameras stop working and you have to reboot.

UltraBlackLinux, 2023-03-26:
since today's libcamera update everything works as expected for me.

Pigpog, 2023-05-04:
The cameras have been working better than ever before for me now. I think this can be closed

closed 2023-05-19 by Danct12