dreemurrs-embedded/Pine64-Arch #504: Can't update image (archlinux-pinephone-phosh-20220729.img.xz )

bfra2373, 2022-12-21:
Hi,

I had issue with my latest update and while I was wondering if it was my config and tryed to test on a brand new image, I could not update/Upgrade the latest image.

- Device: (PinePhone v1.2)
- Kernel Version : $(from the image(I'll try to edit later on)
- UI: phosh

### Steps to reproduce

sudo pacman -Syyu

### Expected behavior

update and upgrade

### Actual behavior

No update or upgrade could be perform

### Logfiles and additional information

```
checking for file conflicts...
linux-megi: /boot/dtbs/allwinner/sun50i-h616-orangepi-zero2.dtb exists in filesystem
linux-megi: /boot/dtbs/allwinner/sun50i-h616-x96-mate.dtb exists in filesystem
linux-megi: /boot/dtbs/rockchip/rk3308-rock-pi-s.dtb exists in filesystem
linux-megi: /boot/dtbs/rockchip/rk3566-quartz64-b.dtb exists in filesystem
linux-megi: /boot/dtbs/rockchip/rk3566-roc-pc.dtb exists in filesystem
linux-megi: /boot/dtbs/rockchip/rk3566-soquartz-cm4.dtb exists in filesystem
linux-megi: /boot/dtbs/rockchip/rk3568-rock-3a.dtb exists in filesystem
Errors occurred, no packages were upgraded.
```

shulamy1, 2023-01-04:
i did it few days ago without problems
try to remove those files
ezik

bfra2373, 2023-01-13:
Indeed!

closed 2023-01-13 by bfra2373