dreemurrs-embedded/Pine64-Arch #506: Unable to use sudo after update

JKakaofanatiker, 2022-12-25:
I just installed Arch and updated it. After a reboot I can't use sudo anymore. It says that a file named `libcrypto.so.1.1` is missing.

Danct12, 2022-12-25:
Yikes, not this again. This is a upstream issue from Arch Linux ARM themselves with OpenSSL being rebuilt. :\

Danct12, 2022-12-25:
I'm not able to reproduce the issue on a fully updated install, can you be more specific on which image and how you update it?

DrRac27, 2022-12-26:
This can also happen if you do a partial upgrade. I had the same problem once as I didn't have enough free storage and the upgrade failed. I managed to fix it via booting from SD card and chrooting into it but if you just installed I would rather reinstall as it's easier.

Danct12, 2023-01-04:
Reporter did not response, and it's assumed to be a partial upgrade (not supported by us) or they powered off the device while an upgrade is going on.

closed 2023-01-04 by Danct12