dreemurrs-embedded/Pine64-Arch #507: mobile-config-firefox-3.1.0-1 causes firefox to ignore touch screen input

dukla2000, 2022-12-29:
- Device: PinePhone Pro
- Kernel Version : 6.0.10-1-danctnix
- UI: sxmo/sway

### Steps to reproduce
Installing mobile-config-firefox moves tabs to bottom of screen, but screen stops responding to touch input.
Removing mobile-config-firefox moves tabs back to top but screen responds to touch input

### Expected behavior
Touch input to work, e.g. clicking on a link

### Actual behavior
Nothing happens when touching a link

### Logfiles and additional information
can provide if necessary but think spurious.

manner82, 2022-12-31:
Have the same issue, I recommend installing desktoplike-mobile-config-firefox from aur instead. That works for now.

https://aur.archlinux.org/packages/desktop-like-mobile-config-firefox

dariox86, 2023-01-01:
Confirmed on regular PinePhone as well.

Danct12, 2023-01-02:
A workaround is available in mobile-config-firefox 3.2.0-1 and is currently in danctnix-testing. Please test.

dariox86, 2023-01-02:
It works for me, but the toolbar is now at the top.

Danct12, 2023-01-02:
Yup, that is the workaround.

dariox86, 2023-01-02:
OK, thank you.

dukla2000, 2023-01-02:
Looks & works fine on my PinePhone Pro, both sxmo/sway & phosh have issue resolved.

closed 2023-01-02 by dukla2000