dreemurrs-embedded/Pine64-Arch #508: Pacman repo not available

gianmarcogg03, 2022-12-31:
When I try to update the system or install a new package it fails because a repo does not work anymore: `p64.arikawa-hi.me` returns a connection refused error.

- Device: PinePhone Pro
- Kernel Version : 5.18.14-1-danctnix
- UI: Phosh

### Steps to reproduce

1. Open the terminal;
2. Update the system or install something with pacman.

### Expected behavior

It should update or install with no issues.

### Actual behavior

It fails to connect to the `p64.arikawa-hi.me` repo (connection refused).

i3Craig, 2022-12-31:
To add some additional information to this, it appears that the domain name `p64.arikawa-hi.me` no longer resolves to an IP address. Thus, `pacman` is unable to install updates. 

[image]

Danct12, 2022-12-31:
We're aware of this issue and currently is setting up a backup mirror.

Danct12, 2022-12-31:
Repository is now back up.

closed 2022-12-31 by Danct12

i3Craig, 2022-12-31:
I can confirm that updates are working now. Thanks for maintaining this distro fixing this server issue quickly!